OpenNuvoton/NUC970_Linux_Kernel #1: SD-card driver is working incorrectly

konachan700, 2017-02-20:
Hello!

I find a bug in SD-card driver. 

When system going to poweroff/reboot, the sd card will shudown too early, before unmounting the FS. It's broke a filesystem. This bug is present on all systems (debian, slakware, buildroot, ...), what used sd-card as root partition. 

It's critical bug, bacause the spi flash too slow and small for many use case. 

`mmcblk0: error -110 sending stop command, original cmd response 0x900, card status 0xd00`

```
[....] All processes ended within 1 seconds...[ ok done.
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
Sending command 13 as 0x9008D03, arg = 0xE6240000, blocks = 0, length = 0
Sending command 12 as 0x9008C03, arg = 0x00000000, blocks = 0, length = 0
mmcblk0: error -110 sending stop command, original cmd response 0xb00, card status 0xb00
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
Sending command 13 as 0x9008D03, arg = 0xE6240000, blocks = 0, length = 0
Sending command 12 as 0x9008C03, arg = 0x00000000, blocks = 0, length = 0
mmcblk0: error -110 sending stop command, original cmd response 0xb00, card status 0xb00 
```

Fill log here: 
http://pastebin.com/1k6xpewX 
http://pastebin.com/mULmYQc3

yachen, 2017-02-21:
Hi,

So the file system is broke because SD is shut down while there're data waiting to be flushed out?
Can you try to issue a sync command before power off/reboot the system?

Sincerely,

Yi-An Chen

konachan700, 2017-02-21:
Yes. 
Yes, i do "sync", it hasn't any effect. 
And when I am mount rootfs as readonly, I see many reading error and stop of poweroff/reboot process. Problem is appears when one of mmc kernel process gets a SIGTERM/SIGKILL/SIGCONT/... signals, e.g. on 
`killall -s SIGKILL -1`

```
root@slackware:~# kill -s SIGCONT -1
root@slackware:~# dd if=/dev/mmcblk0 of=/dev/null bs=1M count=10 skip=17 
Bus error
root@slackware:~# EXT4-fs error (device mmcblk0p3): __ext4_get_inode_loc:4027: inode #21989: block 1521: comm syslogd: unable to read itable block
EXT4-fs (mmcblk0p3): Remounting filesystem read-only
EXT4-fs error (device mmcblk0p3) in ext4_reserve_inode_write:4950: IO failure
EXT4-fs error (device mmcblk0p3) in ext4_dirty_inode:5069: IO failure
EXT4-fs error (device mmcblk0p3): ext4_wait_block_bitmap:466: comm syslogd: Cannot read block bitmap - block_group = 0, block_bitmap = 115
EXT4-fs error (device mmcblk0p3): ext4_discard_preallocations:3935: comm syslogd: Error loading buddy information for 0
```

Problem is present on many different SD-card, it's no broken card.

jury093, 2017-02-21:
Hi!
I have similar bug.
ARM - nuc976.
Kernel version "Linux version 3.10.101 (jury093@ubuntu64) (gcc version 4.3.4 (GCC) )"
bootargs 'console=ttyS0,115200n8 root=/dev/mmcblk0p1 rw rootwait'
Debian-7-armel (generated by standart debootstrap utility).
After "reboot" or "poweroff" and cycle power off/on:
`SDH - Reading 2048 bytes [phy_addr = 0x39e0000]
SDH xfer done.
[ ok done.
[....] All processes ended within 1 seconds...[ ok done.
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
Sending command 13 as 0x9008D03, arg = 0xE6240000, blocks = 0, length = 0
Sending command 12 as 0x9008C03, arg = 0x00000000, blocks = 0, length = 0
mmcblk0: error -110 sending stop command, original cmd response 0xb00, card status 0xb00
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
Sending command 13 as 0x9008D03, arg = 0xE6240000, blocks = 0, length = 0
Sending command 12 as 0x9008C03, arg = 0x00000000, blocks = 0, length = 0
mmcblk0: error -110 sending stop command, original cmd response 0xb00, card status 0xb00
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
Sending command 13 as 0x9008D03, arg = 0xE6240000, blocks = 0, length = 0
Sending command 12 as 0x9008C03, arg = 0x00000000, blocks = 0, length = 0
mmcblk0: error -110 sending stop command, original cmd response 0xb00, card status 0xb00
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
Sending command 13 as 0x9008D03, arg = 0xE6240000, blocks = 0, length = 0
mmcblk0: error -110 sending stop command, original cmd response 0xb00, card status 0x400900
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
end_request: I/O error, dev mmcblk0, sector 98330
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
Sending command 18 as 0x9069203, arg = 0x0001801A, blocks = 6, length = 512
SDH - Reading 3072 bytes [phy_addr = 0x39e0000]
end_request: I/O error, dev mmcblk0, sector 98330
rsyslog: Input/output error  `

Volume can't mount (no table of partitions)..

schung1218, 2017-02-22:
Hi,
I tried to replicate the problem. 
system going to reboot, the file system as root partition is work in sd card .

[https://github.com/schung1218/temp/blob/master/putty.log](https://github.com/schung1218/temp/blob/master/putty.log)

jury093, 2017-02-22:
> I tried to replicate the problem. system going to reboot, the file system as root partition is work in sd card 

Your right, but your system busybox-based:
**BusyBox v1.22.1 (2015-04Sending command 18**
and never use "command 12"..
Problem exist for Debian/Ubuntu with full function rootfs

yachen, 2017-02-22:
Hi,

So you're executing Ubuntu for ARM on NUC976, except the kernel is replaced with the one built from Nuvoton's tree?

Sincerely,

Yi-An Chen

konachan700, 2017-02-22:
Yes. I am tried a ubuntu arm, debian arm, slackware arm, buildroot. I am replaced default kernel to Nuvoton's kernel on all this systems.

yachen, 2017-02-23:
Hi,

So you can reproduce this issue with buildroot as well?
Can you share your buildroot configuration with us? It's more convenient for us to check this issue using buildroot than Ubuntu/Debian...

Sincerely,

Yi-An Chen

yachen, 2017-02-24:
Hi @konachan700 ,

It would be really helpful if you could provide us your buildroot setting.
We have a buildroot framework under testing now but are unable to reproduce the problem. 

Sincerely,

Yi-An Chen

konachan700, 2017-02-28:
[link to BR and kernel configs](https://drive.google.com/drive/folders/0B0XSKpJ18B2sQ05ZWjBYcjBJYjg?usp=sharing)

In BR i don't see any error in dmesg, but reboot process is stopping where SIGTERM sent to all processes. I am share the full BR binary image and configs. Source code is too big for sharing (4GB+ zipped).

jury093, 2017-03-05:
> So you're executing Ubuntu for ARM on NUC976, except the kernel is replaced with the one built from Nuvoton's tree?

yes. I try attach rootfs tarball with Debian-7-armel (aprox 64MB) .. hmm.. "Try again with a file smaller than 10MB"
https://drive.google.com/file/d/0BximTd_xFGq8d3ZrYzhwRERHUXM/view?usp=sharing
download and unpack to free part of SD card. try to load: username - root and password - password.
This rootfs work perfectly with 9260/9g45/am335x/imx6 etc, but crashed with nuc976 kernel after "poweroff" or "reboot" command..

yachen, 2017-03-06:
Hi @jury093,

Thanks. Will test this issue based on the file system you provided.

Sincerely,

Yi-An Chen

yachen, 2017-03-07:
Hi @jury093 ,

After reboot command, I see kernel stop output after following message. But didn't see any I/O error.

Sincerely,

Yi-An Chen

root@armets:~# reboot
[  677.570000] Sending command 18 as 0x9209203, arg = 0x00843420, blocks = 32, length = 512
[  677.580000] SDH - Reading 16384 bytes [phy_addr = 0x3a20000]
[  677.590000] SDH xfer done.
[  677.610000] Sending command 18 as 0x9209203, arg = 0x00842DF0, blocks = 32, length = 512
[  677.610000] SDH - Reading 16384 bytes [phy_addr = 0x3a20000]
[  677.620000] SDH xfer done.
[  677.630000] Sending command 18 as 0x9109203, arg = 0x00842E10, blocks = 16, length = 512
[  677.640000] SDH - Reading 8192 bytes [phy_addr = 0x3a20000]
[  677.640000] SDH xfer done.
[  677.670000] Sending command 18 as 0x9089203, arg = 0x00487078, blocks = 8, length = 512
[  677.680000] SDH - Reading 4096 bytes [phy_addr = 0x3a20000]
[  677.690000] SDH xfer done.
[  677.700000] Sending command 25 as 0x9109903, arg = 0x00485B00, blocks = 16, length = 512
[  677.700000] SDH - Transmitting 8192 bytes
[  677.710000] SDH xfer done.

Broadcast message from root@armets ([  677.720000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  677.730000] Sending command 25 as 0x9109903, arg = 0x00071458, blocks = 16, length = 512

The sys[  677.730000] SDH - Transmitting 8192 bytes
tem is going down for re[  677.740000] SDH xfer done.
boot NOW!
[  677.750000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  677.760000] Sending command 25 as 0x9189903, arg = 0x00488310, blocks = 24, length = 512
[  677.770000] SDH - Transmitting 12288 bytes
[  677.770000] SDH xfer done.
[  677.780000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  677.790000] Sending command 25 as 0x9089903, arg = 0x00402140, blocks = 8, length = 512
[  677.790000] SDH - Transmitting 4096 bytes
[  677.800000] SDH xfer done.
[  677.800000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  677.810000] Sending command 25 as 0x9089903, arg = 0x00482E98, blocks = 8, length = 512
[  677.820000] SDH - Transmitting 4096 bytes
[  677.830000] SDH xfer done.
[  677.830000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  677.840000] Sending command 25 as 0x9809903, arg = 0x00443700, blocks = 128, length = 512
[  677.850000] SDH - Transmitting 65536 bytes
[  677.860000] SDH xfer done.
[  677.860000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  677.870000] Sending command 25 as 0x9089903, arg = 0x00443780, blocks = 8, length = 512
[  677.880000] SDH - Transmitting 4096 bytes
[  677.890000] SDH xfer done.
[  677.890000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  677.910000] Sending command 25 as 0x9089903, arg = 0x00002008, blocks = 8, length = 512
[  677.910000] SDH - Transmitting 4096 bytes
[  677.920000] SDH xfer done.
[  677.920000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  677.930000] Sending command 25 as 0x9089903, arg = 0x00002ED0, blocks = 8, length = 512
[  677.940000] SDH - Transmitting 4096 bytes
[  677.950000] SDH xfer done.
[  677.950000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  677.960000] Sending command 25 as 0x9089903, arg = 0x00402010, blocks = 8, length = 512
[  677.970000] SDH - Transmitting 4096 bytes
[  677.980000] SDH xfer done.
[  677.980000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  678.000000] Sending command 25 as 0x9089903, arg = 0x00486018, blocks = 8, length = 512
[  678.000000] SDH - Transmitting 4096 bytes
[  678.010000] SDH xfer done.
[  678.010000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  678.020000] Sending command 25 as 0x9109903, arg = 0x00802110, blocks = 16, length = 512
[  678.030000] SDH - Transmitting 8192 bytes
[... 358 lines cut ...]
[  682.430000] SDH - Transmitting 4096 bytes
[  682.440000] SDH xfer done.
[  682.440000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  682.450000] Sending command 25 as 0x9189903, arg = 0x008021E0, blocks = 24, length = 512
[  682.460000] SDH - Transmitting 12288 bytes
[  682.460000] SDH xfer done.
[  682.470000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  682.480000] Sending command 25 as 0x9189903, arg = 0x00802200, blocks = 24, length = 512
[  682.490000] SDH - Transmitting 12288 bytes
[  682.490000] SDH xfer done.
[  682.500000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  682.510000] Sending command 25 as 0x9209903, arg = 0x00802238, blocks = 32, length = 512
[  682.520000] SDH - Transmitting 16384 bytes
[  682.520000] SDH xfer done.
[  682.530000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[  682.540000] Sending command 25 as 0x9089903, arg = 0x00C02220, blocks = 8, length = 512
[  682.540000] SDH - Transmitting 4096 bytes
[  682.550000] SDH xfer done.
[  682.560000] Sending command 13 as 0x9018D03, arg = 0x59B40000, blocks = 0, length = 0
[....] Asking all remaining processes to terminate...[  682.650000] Sending command 18 as 0x9209203, arg = 0x00842240, blocks = 32, length = 512
[  682.650000] SDH - Reading 16384 bytes [phy_addr = 0x3a20000]
[  682.660000] SDH xfer done.
[  682.670000] Sending command 18 as 0x9089203, arg = 0x00842260, blocks = 8, length = 512
[  682.680000] SDH - Reading 4096 bytes [phy_addr = 0x3a20000]
[  682.680000] SDH xfer done.

jury093, 2017-03-07:
Hi yachen!
1. please, try make 2-3 cycles to boot system and "poweroff" command
2. what flash card your used? I use uSD (2GB, 4GB and 8GB)..
Thank you

yachen, 2017-03-07:
Hi @jury093 ,

1. Ok
2. I'm using 8GB micro SD.

Sincerely,

Yi-An Chen

yachen, 2017-03-08:
Hi @jury093 ,

I've tested more than 50 times yet cannot reproduce the problem.
There are some debug messages were comment out ( //nuc970_sd_debug()) in driver,  could you un-comment them and log the console message after issue poweroff command for me? Thanks.

Sincerely,

Yi-An Chen

jury093, 2017-03-09:
Hi yachen!
sorry for delay - local holyday :)
I get fresh kernel "git clone https://github.com/OpenNuvoton/NUC970_Linux_Kernel.git"
new uSD, ext4, kernel with debug output for mmc, boot
full log at http://pastebin.com/xPx0qWJB

yachen, 2017-03-09:
Hi,

Did you un-comment the//nuc970_sd_debug() in driver?
I cannot find and message print out from line 584.
//nuc970_sd_debug("Event = 0x%0X [0x%08X 0x%08X] <0x%x>\n", status, cmd->resp[0], cmd->resp[1], err); 

Sincerely,

Yi-An Chen

jury093, 2017-03-09:
Hi yachen!
yes, I uncomment for debug.. sorry for long log
```
U-Boot 2013.04-rc2 (Sep 21 2016 - 14:01:21)

CPU: NUC976
DRAM:  64 MiB
MMC:   mmc: 0
SF: Detected MX25L12805D with page size 64 KiB, total 16 MiB
In:    serial
Out:   serial
Err:   serial
Net:   emac
Hit any key to stop autoboot:  3  0 
U-Boot> run mmc_boot
mmc0 is current device
2133016 bytes read in 303 ms (6.7 MiB/s)
## Booting kernel from Legacy Image at 00300000 ...
   Image Name:   Linux-3.10.101+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2132952 Bytes = 2 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
SF: Detected MX25L12805D with page size 64 KiB, total 16 MiB
   Loading Kernel Image ... OK
OK

Starting kernel ...

Uncompressing Li done, booting tBooting Linux on physical CPU 0x0
Initializing cgroup subsys cpuset
Initializing cgroup subsys cpu
Initializing cgroup subsys cpuacct
Linux version 3.10.101+ (jremizov@diehard) (gcc version 4.3.4 (GCC) ) #3 PREEMPT Thu Mar 9 12:42:37 MSK 2017
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC970
Memory policy: ECC disabled, Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: console=ttyS0,115200n8 root=/dev/mmcblk0p1 rw rootwait
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 64MB = 64MB total
Memory: 60432k/60432k available, 5104k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    vmalloc : 0xc4800000 - 0xff000000   ( 936 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03dbf94   (3920 kB)
      .init : 0xc03dc000 - 0xc03fb11c   ( 125 kB)
      .data : 0xc03fc000 - 0xc042ff40   ( 208 kB)
       .bss : 0xc042ff40 - 0xc0460250   ( 193 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Preemptible hierarchical RCU implementation.
NR_IRQS:625
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 4294967286ms
Console: colour dummy device 80x30
console [ttyS0] enabled
Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 512
Initializing cgroup subsys devices
Initializing cgroup subsys freezer
Initializing cgroup subsys perf_event
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0xc031c500 - 0xc031c53c
[... 2551 lines cut ...]
Sending command 18 as 0x9089203, arg = 0x0B1AB000, blocks = 8, length = 512
SDH xfer done.
SDH - Reading 4096 bytes [phy_addr = 0x39c0000]
Sending command 18 as 0x9089203, arg = 0x0B1AB000, blocks = 8, length = 512
SDH xfer done.
SDH - Reading 4096 bytes [phy_addr = 0x39c0000]
Sending command 18 as 0x9089203, arg = 0x0B1AB000, blocks = 8, length = 512
SDH xfer done.
SDH - Reading 4096 bytes [phy_addr = 0x39c0000]
halt: Input/output error
SDH xfer done.
Sending command 25 as 0x9809903, arg = 0x302CE000, blocks = 128, length = 512
SDH - Transmitting 65536 bytes
Sending command 13 as 0x9018D03, arg = 0x85E90000, blocks = 0, length = 0
Sending command 25 as 0x9089903, arg = 0x302DE000, blocks = 8, length = 512
SDH - Transmitting 4096 bytes
Sending command 13 as 0x9018D03, arg = 0x85E90000, blocks = 0, length = 0
[....] startpar: service(s) returned failure: rsyslog hwclock.sh umountnfs.sh networking umountfs umountroot halt ...[FAIL failed!
INIT:аno more processes left in this runlevel
SDH xfer done.
Sending command 25 as 0x9789903, arg = 0x302DF000, blocks = 120, length = 512
SDH - Transmitting 61440 bytes
Sending command 13 as 0x9018D03, arg = 0x85E90000, blocks = 0, length = 0 
```
this last string from console

jury093, 2017-03-09:
Oh! I stupid :)
see new log for debug and uncommented 584 string:
http://pastebin.com/HPxwaPe8

yachen, 2017-03-10:
Many thanks jury093,
Our engineer find some incorrect SD state change in the log and is checking the root cause.

yachen, 2017-03-16:
Hi @jury093,

Can you modify nuc970_sd_completed_command() and test again?
We guess the take was interrupted by signal instead of transfer complete thus cause the state error.

        if (data)
        {
-           wait_event_interruptible(sd_wq_xfer, (sd_state_xfer != 0))
+            if (wait_event_interruptible(sd_wq_xfer, (sd_state_xfer != 0)))
+			{
+				nuc970_sd_debug("\ndata: interrupted by signal !!!\n\n");
+				return -ERESTARTSYS;
+			}
        }

Thanks,

Yi-An Chen

jury093, 2017-03-16:
Hi Yi-An Chen!
I can make this test only tommorow, sorry..

jury093, 2017-03-17:
Hi Yi-An Chen!
I make modification for **nuc970_sd_completed_command()** function:
`//        if (data)
//        {
//            wait_event_interruptible(sd_wq_xfer, (sd_state_xfer != 0));
//        }
        if (data)
        {
            wait_event_interruptible(sd_wq_xfer, (sd_state_xfer != 0));
            if(wait_event_interruptible(sd_wq_xfer, (sd_state_xfer != 0)))
		{
		nuc970_sd_debug("\ndata: interrupted by signal !!!\n\n");
		return -ERESTARTSYS;
		}
        }`
rebuild with warning:
`  CALL    scripts/checksyscalls.sh
  GEN     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/file2alias.o
  HOSTLD  scripts/mod/modpost
  CHK     include/generated/compile.h
  CC      drivers/mmc/host/nuc970_sd.o
drivers/mmc/host/nuc970_sd.c: In function 'nuc970_sd_completed_command':
drivers/mmc/host/nuc970_sd.c:615: warning: 'return' with a value, in function returning void`
and boot from mmc (until boot I run e2fsck for clear mmcblk0p1 for errors)
Full log:
http://pastebin.com/24qzmTS5
after "poweroff" command system is make shutdown to strings:
`data: interrupted by signal !!!
SDH xfer done.`
and freezed. But for normal shutdown I can't "Will now halt" message and insert this uSD to USB Flash reader I check filesystem:
`~# e2fsck /dev/sda1
e2fsck 1.42.12 (29-Aug-2014)
/dev/sda1: recovering journal
Setting free blocks count to 385353 (was 385562)
/dev/sda1: clean, 13819/118800 files, 89391/474744 blocks`

yachen, 2017-03-17:
Hi @jury093 ,

This log confirms the root cause of file system corruption. The wait_event_interruptible() was waked by a signal instead of transfer complete interrupt which the design is expect. So the following transfer are corrupted due to state error. We'll check how to recover from this state. Maybe  replace it with a wait_event_timeout() so the transfer will not interrupt by signal while the timeout guarantees the code won't block forever,

Sincerely,

Yi-An Chen

jury093, 2017-11-24:
Hi Yachen!
A week ago I clone kernel sources from https://github.com/OpenNuvoton/NUC970_Linux_Kernel, build uImage and boot with Debian 7 armel rootFS..
I make test for 3 card:
**Kingston 2GB uSD
Transcend 2GB uSD
SunDisk Ultra 8GB uSD HCI**
No errors after ~20 cycles reboot/powerdown each of these cards.
I think,  this bug (distruction rootfs/partition after reboot/powerdown) can be close
Thank

closed 2017-11-27 by yachen