OpenNuvoton/NUC970_Linux_Kernel #2: usb gadget with cdc ethernet not work

junwushi, 2017-03-04:
rndis feature is not work,  show linux USB Ehternet/RNDIS gadget #4 in window device tree ,but it is with "！“，does not work.

junwushi, 2017-03-04:
if building kernel with usb gadget->serial gadget(with cdc acm and cdc obex support), the serial comport device will often not work,many times reboot the linux device ,the comport device will work in window xp/10.

yachen, 2017-03-06:
Hi @junwushi ,

We are aware of the RNDIS issue and have engineer checking it now.
Can you provide us your kernel config for CDC issue? We've test it before and have other customers using it but didn't see the issue you report. Thanks. 

Sincerely,

Yi-An Chen

junwushi, 2017-03-06:
Dear Chen,
    i am already upload my .config,please remove the extension ".zip" and  decompress it with TAR.
[config.tar.zip](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/819978/config.tar.zip)

yachen, 2017-03-06:
Hi @junwushi ,

I'm able to reproduce the CDC unstable issue and have forwarded the issue to the engineer who maintain the NUC970 gadget driver. Will update the fix to Github as soon as it's fixed.

Sincerely,

Yi-An Chen

yachen, 2017-03-17:
Hi @junwushi ,

Please check if commit c1a9947108e2929c7621f7b91fa3aaf4d586b1eb fix CDC issue or not.
RNDIS issue is still under investigating.

Sincerely,

Yi-An Chen

junwushi, 2017-03-20:
rndis still not work, device driver install  normal,but network device always show in disable status,can't enable even you maunul enable it. the attach is the rndis driver for windows.
[linux.rar.zip](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/854522/linux.rar.zip)

yachen, 2017-03-20:
Hi @junwushi ,

This patch is intent for fix the "usb gadget->serial gadget serial comport device will often not work" issue you reported. Please let us know if this fixed the CDC issue.  Thanks.
We're still working on the RNDIS issue and will push to this repository after it's fixed.

Sincerely,

Yi-An Chen

junwushi, 2017-03-20:
yes, the CDC seems stable,will observe it longer time.

junwushi, 2017-03-22:
I think cdc issue not sloved.
1.after I update  the c1a9947  the system often crashed, not show any crash information in debug, only hang up. and sometimes the sytem is startup ok, below is the debug message when the system hangup . but it is not hang up before update.
Starting kernel ...

Uncompressing Li done, booting t[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.1011+ (jwshi@leo-VirtualBox) (gcc version 6.2.0 (GCC) ) #72 PREEMPT Tue Mar 21 19:44:53 CST 2017
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: NUC970
[    0.000000] cma: CMA: reserved 4 MiB at 03c00000
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rw rootflags=inband-tags console=ttyS0,115200n8 init=/sbin/init mem=64M lpj=744448 vt.global_cursor_default=0
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 64MB = 64MB total
[    0.000000] Memory: 55448k/55448k available, 10088k reserved, 0K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     vmalloc : 0xc4800000 - 0xff000000   ( 936 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc04a9f64   (4744 kB)
[    0.000000]       .init : 0xc04aa000 - 0xc04ce930   ( 147 kB)
[    0.000000]       .data : 0xc04d0000 - 0xc0514400   ( 273 kB)
[    0.000000]        .bss : 0xc0514400 - 0xc053c8cc   ( 162 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:625
[    0.000000] sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 4294967286ms
[    0.000000] Console: colour dummy device 80x30
[    0.000000] console [ttyS0] enabled
[    0.140000] Calibrating delay loop (skipped) preset value.. 148.88 BogoMIPS (lpj=744448)
[    0.150000] pid_max: default: 32768 minimum: 301
[    0.150000] Mount-cache hash table entries: 512
[    0.160000] Initializing cgroup subsys devices
[    0.170000] Initializing cgroup subsys freezer
[    0.170000] Initializing cgroup subsys perf_event
[    0.170000] CPU: Testing write buffer coherency: ok
[    0.180000] Setting up static identity map for 0xc03ac570 - 0xc03ac5c8
[    0.190000] devtmpfs: initialized
[    0.280000] pinctrl core: initialized pinctrl subsystem
[    0.290000] NET: Registered protocol family 16
[    0.300000] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.480000] bio: create slab <bio-0> at 0
[    0.490000] SCSI subsystem initialized
[    0.500000] usbcore: registered new interface driver usbfs
[    0.500000] usbcore: registered new interface driver hub
[    0.510000] usbcore: registered new device driver usb
[    0.520000] i2c_gpio_probe - pdev = i2c-gpio
[    0.520000] platform i2c-gpio.2: Driver i2c-gpio requests probe deferral
[    0.530000] pps_core: LinuxPPS API ver. 1 registered
[    0.540000] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.550000] PTP clock support registered
[    0.560000] FS-Cache: Loaded
[    0.560000] CacheFiles: Loaded
[    0.670000] NET: Registered protocol family 2
[    0.680000] TCP established hash table entries: 512 (order: 0, 4096 bytes)
[    0.690000] TCP bind hash table entries: 512 (order: -1, 2048 bytes)
[    0.700000] TCP: Hash tables configured (established 512 bind 512)
[    0.700000] TCP: reno registered
[    0.710000] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.710000] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.720000] NET: Registered protocol family 1
[    0.720000] NetWinder Floating Point Emulator V0.97 (double precision)
[... 102 lines cut ...]
[    1.810000] l2tp_ip6: L2TP IP encapsulation support for IPv6 (L2TPv3)
[    1.820000] NET: Registered protocol family 36
[    1.840000] g_serial gadget: high-speed config #2: CDC ACM config
[    1.850000] Key type encrypted registered
[    1.850000] i2c_gpio_probe - pdev = i2c-gpio
[    2.160000] input: ft5x0x_ts as /devices/platform/i2c-gpio.2/i2c-2/2-0038/input/input0
[    2.330000] ft5x0x_ts 2-0038: Screen size: 480X272 initialized:  IRQ pin 174 ==> 430, Reset pin 173.
[    2.340000] i2c-gpio i2c-gpio.2: using pins 193 (SDA) and 192 (SCL)
[    2.340000] nuc970-rtc nuc970-rtc: setting system clock to 2016-10-26 12:58:16 UTC (1477486696)
[    2.350000] yaffs: dev is 32505858 name is "mtdblock2" rw
[    2.360000] yaffs: passed flags "inband-tags"
[    5.590000] VFS: Mounted root (yaffs2 filesystem) on device 31:2.
[    5.600000] devtmpfs: mounted
[    5.600000] Freeing unused kernel memory: 144K (c04aa000 - c04ce000)
INIT: version 2.88 bootin
Starting udev
[    6.750000] udevd[669]: starting version 3.2
[    6.880000] udevd[670]: starting eudev-3.2

2.when the sytem up, login console, then reconnect the usb cable, will effect console rs232 port and ,console will not response any type, after a long time console display blow:
[    32.4000000]ttyS0: 1 input overrubs(s)
>
>
>
>

junwushi, 2017-03-25:
the issue is still there, usually all seems ok(device is ok, port is ready). but just can't comunicate,type any char not response in ttyGS0(getty). after reboot many times ,it is restore working. why?

yachen, 2017-03-27:
Hi @junwushi ,

May I know how do you test the ttyGS0? I'm testing with putty on my Windows machine and cannot reproduce the problem so far.

Sincerely,

Yi-An Chen

junwushi, 2017-03-30:
if application  open the ttyGS0 ,then  disconnect the usb cable,the ttyGS0 device lost in linux,but the application  maybe hang that still  perform some function of the device ttyGS0 (like selct() etc.). and  window side does not know the comport is invalid,  not received any notify for disconnect.

junwushi, 2017-03-30:
other thing,the battery adc reference voltage use  #define INTERNAL_VALTAGE 2500 in bsp adc driver,but use 3300 is corrent ADC data result in fact . why? please confirm my supcisions are corrent? i am not sure,and the program  guide docment is so simple.

schung1218, 2017-03-30:
Hi @junwushi 
The battery voltage detection select internal voltage(2.5v)  or external reference pin.
if the external reference pin connect 3.3v, use 3300 is correct.
if the external reference pin is floating , use 2500 is correct.

junwushi, 2017-03-31:
sometime hang in udc drive line:
while ((_raw_read(comtroller.reg + REG_USBD_IRQ_STAT) &0x20 )==0 );
when if you muilttime plus on/off the cable and reopen ttyGS0 for comunucation.

junwushi, 2017-04-01:
and I found the USB_FLT_DET irq never execute,not print plug in and un plug  message in console.

yachen, 2017-04-07:
@junwushi ,

What board are you testing? "USB_FLT_DET irq never execute" should be hardware issue.

Sincerely,

Yi-An Chen

junwushi, 2017-04-13:
it is host&device otg mode, the ph0 pin always high voltage,becase usb VCC it is connect battery. how to design usb0 in otg mode and ph0 can indentify device plug in and off?

yachen, 2017-04-13:
Hi @junwushi ,

Are you using Nuvoton's DEV board? If so, please set SW34:1 to OFF position and change R83 to 20K Ohm, R84 to 39K Ohm.

Sincerely,

Yi-An Chen

junwushi, 2017-07-24:
after update kernal 107,there are cdc serial report below message:

gateway /dev/ttyS0

gateway login: [   14.310000] g_serial gadget: high-speed config #2: CDC ACM config
Hangup
root
root@gateway:~# [   19.430000] g_serial gadget: high-speed config #2: CDC ACM config
Hangup
[   20.210000] g_serial gadget: high-speed config #2: CDC ACM config
Hangup
[   28.510000] g_serial gadget: high-speed config #2: CDC ACM config
Hangup
[   35.940000] g_serial gadget: high-speed config #2: CDC ACM config
Hangup
[   54.930000] g_serial gadget: high-speed config #2: CDC ACM config
Hangup

junwushi, 2017-11-15:
please this function "static u32 udc_transfer(struct nuc970_ep *ep, u8* buf, size_t size, u32 mode)" in nuc970_udc.c, the function always return 0, it is right?

junwushi, 2017-11-15:
my device usually die and show  the "hangup1" that add by my self for debug ,why? 

write_packet(struct nuc970_ep *ep, struct nuc970_request *req)
{
	struct nuc970_udc *udc = &controller;
	unsigned len, tmp;
	u8  *buf;
	u32 data, i;
	u32 max;

	buf = req->req.buf + req->req.actual;
	prefetch(buf);
//	if (udc->gadget.speed == USB_SPEED_FULL)
//		udelay(500);

	if (ep->ep_num == 0)
	{ //ctrl pipe don't use DMA
		max = ep->ep.maxpacket;
		len = min(req->req.length - req->req.actual, max);

		if (len == 0)
		{
			if (req->req.zero&&!req->req.length)
				__raw_writel(CEP_ZEROLEN, controller.reg + REG_USBD_CEP_CTRL_STAT);
		}
		else
		{
			i=0;
			while ((__raw_readl(controller.reg + REG_USBD_CEP_IRQ_STAT) & 0x1000) != 0x1000)
			{
				i++;
				if (i>10000) printk("hang up1\r\n");
			}
				;

yachen, 2017-11-23:
Hi @junwushi ,

The loop is waiting for previous transfer to complete. Shouldn't stuck there. Did you make other modification as well?
Anyway, please pull latest driver, we fixed some issues including the udc_transfer() always return 0 you reported.

Sincerely,

Yi-An Chen

junwushi, 2017-11-23:
the udc_transfer() always return 0 maybe right, becase i check that code in the dma irq process to increase the count. so don't need return send bytes.
the last code i am checking

junwushi, 2017-11-23:
"udc_transfer() always return 0" is not issue, please change back it, that is right, now test the new code ,that not working, and change back  udc_transfer()  return 0 that working.

yachen, 2017-11-23:
Hi @junwushi ,

Updated, please pull again. Many thanks for the feedback.

Sincerely,

Yi-An Chen

junwushi, 2017-11-23:
ok, working

junwushi, 2017-11-24:
still same issue in udc driver. write_packet only die in while {} section and wait for every time.
other things, below code, first unlock and then lock is right?
static void nuke (struct nuc970_udc *udc, struct nuc970_ep *ep)
{
	while (!list_empty (&ep->queue))
	{
		struct nuc970_request *req;
		req = list_entry (ep->queue.next, struct nuc970_request, queue);
		list_del_init (&req->queue);
		req->req.status = -ESHUTDOWN;
		spin_unlock (&udc->lock);
		req->req.complete (&ep->ep, &req->req);
		spin_lock (&udc->lock);
	}
}

junwushi, 2017-11-30:
now i used the USB cable that maybe not good, and then print below message cycle:
[  309.680000] g_serial gadget: high-speed config #2: CDC ACM config
[  414.200000] g_serial gadget: high-speed config #2: CDC ACM config
[  416.310000] g_serial gadget: high-speed config #2: CDC ACM config
[  438.370000] g_serial gadget: high-speed config #2: CDC ACM config
[  442.360000] g_serial gadget: high-speed config #2: CDC ACM config
[  452.400000] g_serial gadget: high-speed config #2: CDC ACM config
[  484.570000] g_serial gadget: high-speed config #2: CDC ACM config
[  528.790000] g_serial gadget: high-speed config #2: CDC ACM config
[  530.830000] g_serial gadget: high-speed config #2: CDC ACM config
[  534.840000] g_serial gadget: high-speed config #2: CDC ACM config
[  536.830000] g_serial gadget: high-speed config #2: CDC ACM config
[  657.340000] g_serial gadget: high-speed config #2: CDC ACM config
[  657.600000] g_serial gadget: high-speed config #2: CDC ACM config
[  657.870000] g_serial gadget: high-speed config #2: CDC ACM config
[  673.170000] g_serial gadget: high-speed config #2: CDC ACM config
[  673.450000] g_serial gadget: high-speed config #2: CDC ACM config
[  673.730000] g_serial gadget: high-speed config #2: CDC ACM config
[  699.050000] g_serial gadget: high-speed config #2: CDC ACM config
[  699.310000] g_serial gadget: high-speed config #2: CDC ACM config
[  764.210000] g_serial gadget: high-speed config #2: CDC ACM config
[  764.470000] g_serial gadget: high-speed config #2: CDC ACM config
[  789.890000] hangup1!
[  799.800000] g_serial gadget: high-speed config #2: CDC ACM config
[  800.040000] g_serial gadget: high-speed config #2: CDC ACM config
[  800.330000] g_serial gadget: high-speed config #2: CDC ACM config
[  800.580000] g_serial gadget: high-speed config #2: CDC ACM config
[  800.870000] g_serial gadget: high-speed config #2: CDC ACM config
[  801.170000] g_serial gadget: high-speed config #2: CDC ACM config

yachen, 2017-12-01:
We think the lock/unlock could be removed in nuke(). Will do some more test before release update.

junwushi, 2017-12-07:
new issuse:
i connect the 3g module in usb port, it is show below error,and i am be care of to check hardware, not found any problem:

[    2.710000] usb 2-2: new low-speed USB device number 2 using nuc970-ohci
[    2.900000] usb 2-2: device descriptor read/64, error -62
[    3.190000] usb 2-2: device descriptor read/64, error -62
[    3.480000] usb 2-2: new low-speed USB device number 3 using nuc970-ohci
[    3.670000] usb 2-2: device descriptor read/64, error -62
[    3.960000] usb 2-2: device descriptor read/64, error -62
[    4.250000] usb 2-2: new low-speed USB device number 4 using nuc970-ohci

yachen, 2017-12-07:
Hi @junwushi ,

This is a USB signal quality issue. We have other customer's hardware got same error.
First the dongle was misidentified as low speed device, and then driver report error -62.

Sincerely,

Yi-An Chen

yachen, 2018-01-03:
Hi @junwushi ,

Please pull 737c84e and see if it fix the write_packet() loop forever issue.
We saw similar issue while connecting with some PC, and this patch fixed it.

Sincerely,

Yi-An Chen

yachen, 2018-01-18:
Close unless there's further discussion on this issue.

closed 2018-01-18 by yachen

swex, 2019-08-14:
ethernet/RNDIS still not working, any update on this?