OpenNuvoton/NUC970_Linux_Kernel #3: rtc interrupt issue

jury093, 2017-03-06:
Hi all!
load for not pathed driver:
```
[....] Starting periodic command scheduler: cron. ok 
[....] Starting OpenBSD Secure Shell server: sshd. ok 

Debian GNU/Linux 7 nuc976 ttyS0

nuc976 login: root
Password: 
Last login: Sun Mar  5 18:08:20 MSK 2017 from 192.168.0.107 on pts/0
Linux nuc976 3.10.101+ #1 PREEMPT Mon Mar 6 12:01:32 MSK 2017 armv5tejl

root@nuc976:~# hwclock -w
hwclock: select() to /dev/rtc0 to wait for clock tick timed out: No such file or directory

root@nuc976:~# ls /dev/rtc0 -l
crw------- 1 root root 254, 0 Jan  1 03:02 /dev/rtc0

root@nuc976:~# cat /proc/interrupts 
           CPU0       
 15:          0         -  nuc970rtc
 16:       9661         -  nuc970-timer0
 19:        364         -  nuc970-emac0
 21:          5         -  nuc970-emac0
 23:       3979         -  ehci_hcd:usb1
 24:          0         -  ohci_hcd:usb2
 27:          0         -  mmc0
 36:        306         -  ttyS0
 53:          0         -  nuc970-i2c0
Err:          0

root@nuc976:~# poweroff 

Broadcast message from root@nuc976 (ttyS0) (Sat Jan  1 03:10:04 2005):

The system is going down for system halt NOW!
INIT: Switching to runlevel: 0
INIT: Sending processes the TERM signal
[info] Using makefile-style concurrent boot in runlevel 0.
[....] Asking all remaining processes to terminate...done.
[....] All processes ended within 1 seconds...done.
[....] Stopping enhanced syslogd: rsyslogd. ok 
[info] Saving the system clock.
hwclock: select() to /dev/rtc0 to wait for clock tick timed out: Success
```
========================================================================
patch to enable interrupts:

```
drivers/rtc/rtc-nuc970.c

static int __init nuc970_rtc_probe(struct platform_device *pdev)
...
                return -EBUSY;
        }

        #ifdef CONFIG_ENABLE_RTC_WAKEUP
        __raw_writel((1<<24) | __raw_readl(REG_WKUPSER),REG_WKUPSER);
        enable_irq_wake(nuc970_rtc->irq_num);
        #endif

+       rtc_reg_write(nuc970_rtc, REG_RTC_RIER, TICKINTENB);
        return 0;
```
=======================================================================
work now:
```
Debian GNU/Linux 7 nuc976 ttyS0

nuc976 login: root
Password: 
Last login: Sat Jan  1 03:03:26 MSK 2005 on ttyS0
Linux nuc976 3.10.101+ #2 PREEMPT Mon Mar 6 12:21:33 MSK 2017 armv5tejl

root@nuc976:~# hwclock -w
root@nuc976:~# cat /proc/interrupts 
           CPU0       
 15:         81         -  nuc970rtc
 16:       8193         -  nuc970-timer0
 19:        286         -  nuc970-emac0
 21:          5         -  nuc970-emac0
 23:       3958         -  ehci_hcd:usb1
 24:          0         -  ohci_hcd:usb2
 27:          0         -  mmc0
 36:        314         -  ttyS0
 53:          0         -  nuc970-i2c0
Err:          0
```

yachen, 2017-03-07:
Hi,

Thanks for the feedback. But your patch will keep system wake-up from power down mode every second.
We'll post a update allowing user to configure between alarm and tick wake-up.

Sincerely,

Yi-An Chen

jury093, 2017-03-07:
Hi yachen!
May be before enter "power mode" (or suspend?) driver should be disable this interrupt and enable after wake up from deep sleep..
For example: drivers into **driver/rtc** with suspend function (rtc-s3c.c, rtc-at91sam9.c etc) consist "CONFIG_PM_SLEEP"
If I'm not mistaken..
My "patch" little wrong - really need read register, set bit and write register again. don't touch other bits..

yachen, 2017-03-07:
Hi @jury093 ,

We'll open an option for user to enable tick interrupt as wake up source (Default is disable).
And yes, should be a read/modify/write. Wait for couple of minutes, the patch has been submitted into our internal server.

Sincerely,

Yi-An Chen

yachen, 2017-03-07:
Fixed in 1cbc239c368994026db14fd62cd5789a7bf9d052

closed 2017-03-07 by yachen