OpenNuvoton/NUC970_Linux_Kernel #6: keypad can not wakup the system?

junwushi, 2017-05-29:
i am configure with "CONFIG_NUC970_KEYPAD_WKUP " option, and after I  input the command "echo freeze > /sys/power/state " or  "echo mem > /sys/power/state " to suspend, but the system can't be wakeup whatever  I press  any key  . and use keypad to wakeup that need some additional hardware design? see attach file for kernel conflagration.

[config.tar.gz](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/1034891/config.tar.gz)

root@gateway:~# echo freeze > /sys/power/state 
[   18.060000] PM: Syncing filesystems ... done.
[   18.100000] Freezing user space processes ... (elapsed 0.02 seconds) done.
[   18.140000] Freezing remaining freezable tasks ... (elapsed 0.02 seconds) done.
[   18.160000] Suspending console(s) (use no_console_suspend to debug)

junwushi, 2017-06-02:
any advice or respond?

yachen, 2017-06-02:
Hi @junwushi ,

Our engineer is checking this issue.

Sincerely,

Yi-An Chen

mjchen1, 2017-06-06:
please connect row pin to GND or pull low.
then try again.

junwushi, 2017-06-08:
hi@mjchecn1
   your means the row pin always connect to GND ,or when press a key to connect to GND. can you explain it more detail or give me sample of the SCH?, i am try after suspend, and then connect one row to GND, that's not working.

mjchen1, 2017-06-08:
you need connect row pin to pull-low on your PCB,
then enter to suspend, and then press kpi key, that,s can wake-up your system.

mjchen1, 2017-06-08:
because some row pin then same with powe-on setting pin and power-on setting pin will be pull-high.
if row pin be pull-high, kpi will detect key fail.
so, you need pull-low row pin, then kpi wakeup function can work normal.

junwushi, 2017-06-09:
hi@mjchen1,
can this feature implemented at demo board nhs-972-1-cy-1m54? I try at it but wakeup not working. and I connect row2 pin to GND and press k20 at my board but still not working, the attach is our keypad design. is it  one row pin that wakeup key should be pull low or all row pins should be pull low?
[image]
[image]

mjchen1, 2017-06-12:
yes, you need select " pull-up nuc970 matrix keypad pin" in menu config

junwushi, 2017-06-12:
but what's the problem in above design, that can not wakeup.

mjchen1, 2017-06-12:
could you try the config,
I can wake-up my system from KPI.

[20170612_.zip](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/1067213/20170612_.zip)

[image]

junwushi, 2017-06-12:
it is running demo board?

junwushi, 2017-06-12:
now after wakeup, and show below error:
[  139.340000] PM: Syncing filesystems ... done.
[  139.350000] Freezing user space processes ... (elapsed 0.02 seconds) done.
[  139.370000] Freezing remaining freezable tasks ... (elapsed 0.02 seconds) done.
[  139.390000] Suspending console(s) (use no_console_suspend to debug)
[  139.400000] nuc970fb suspend
[  139.400000] BUG: scheduling while atomic: sh/784/0x00000002
[  139.400000] Modules linked in:
[  139.400000] CPU: 0 PID: 784 Comm: sh Tainted: G        W    3.10.101-00048-g98dd3f7ab3a-dirty #19
[  139.400000] Backtrace: 
[  139.400000] [<c00122c0>] (dump_backtrace+0x0/0x108) from [<c00124d0>] (show_stack+0x18/0x1c)
[  139.400000]  r7:c360afa0 r6:c05c35a8 r5:c36a2000 r4:00000000
[  139.400000] [<c00124b8>] (show_stack+0x0/0x1c) from [<c046a888>] (dump_stack+0x20/0x28)
[  139.400000] [<c046a868>] (dump_stack+0x0/0x28) from [<c0046828>] (__schedule_bug+0x48/0x60)
[  139.400000] [<c00467e0>] (__schedule_bug+0x0/0x60) from [<c046c458>] (__schedule+0x39c/0x488)
[  139.400000]  r5:c36a2000 r4:ffffc338
[  139.400000] [<c046c0bc>] (__schedule+0x0/0x488) from [<c046c57c>] (schedule+0x38/0x78)
[  139.400000] [<c046c544>] (schedule+0x0/0x78) from [<c046af88>] (schedule_timeout+0x110/0x1a0)
[  139.400000] [<c046ae78>] (schedule_timeout+0x0/0x1a0) from [<c046c740>] (wait_for_common+0x90/0x1bc)
[  139.400000] [<c046c6b0>] (wait_for_common+0x0/0x1bc) from [<c046c8e4>] (wait_for_completion_interruptible_timeout+0x14/0x18)
[  139.400000]  r8:c05c03fc r7:00000000 r6:00000002 r5:c340a254 r4:c340a000
[  139.400000] [<c046c8d0>] (wait_for_completion_interruptible_timeout+0x0/0x18) from [<c01f9134>] (nuc970fb_suspend+0x40/0x60)
[  139.400000] [<c01f90f4>] (nuc970fb_suspend+0x0/0x60) from [<c0241f24>] (platform_pm_suspend+0x58/0x5c)
[  139.400000]  r5:c0241ecc r4:c05c03c8
[  139.400000] [<c0241ecc>] (platform_pm_suspend+0x0/0x5c) from [<c0246dbc>] (dpm_run_callback.constprop.2+0x20/0x50)
[  139.400000] [<c0246d9c>] (dpm_run_callback.constprop.2+0x0/0x50) from [<c0247a50>] (__device_suspend+0xd0/0x228)
[  139.400000]  r5:c061d4ac r4:c05c03c8
[  139.400000] [<c0247980>] (__device_suspend+0x0/0x228) from [<c024828c>] (dpm_suspend+0x64/0x228)
[  139.400000]  r9:00000000 r8:c05e3688 r7:c061d4ac r6:c05e3688 r5:c05c03c8
[  139.400000] r4:c05c041c
[  139.400000] [<c0248228>] (dpm_suspend+0x0/0x228) from [<c02486b4>] (dpm_suspend_start+0x60/0x68)
[  139.400000] [<c0248654>] (dpm_suspend_start+0x0/0x68) from [<c004f4a8>] (suspend_devices_and_enter+0x50/0x354)
[  139.400000]  r5:c0610338 r4:00000003
[  139.400000] [<c004f458>] (suspend_devices_and_enter+0x0/0x354) from [<c004f950>] (pm_suspend+0x1a4/0x208)
[  139.400000] [<c004f7ac>] (pm_suspend+0x0/0x208) from [<c004e808>] (state_store+0x78/0xdc)
[  139.400000]  r5:00000003 r4:c05c3f30
[  139.400000] [<c004e790>] (state_store+0x0/0xdc) from [<c01c1844>] (kobj_attr_store+0x1c/0x28)
[  139.400000] [<c01c1828>] (kobj_attr_store+0x0/0x28) from [<c012370c>] (sysfs_write_file+0xe0/0x19c)
[  139.400000]  r5:c35980c0 r4:00000004
[  139.400000] [<c012362c>] (sysfs_write_file+0x0/0x19c) from [<c00c62f0>] (vfs_write+0xb8/0x198)
[  139.400000] [<c00c6238>] (vfs_write+0x0/0x198) from [<c00c68ac>] (SyS_write+0x44/0x80)
[  139.400000]  r9:00000004 r8:00000000 r7:00000000 r6:00000000 r5:01af1040
[  139.400000] r4:c3783be0
[  139.400000] [<c00c6868>] (SyS_write+0x0/0x80) from [<c000ef00>] (ret_fast_syscall+0x0/0x34)
[  139.400000]  r9:c36a2000 r8:c000f0a8 r7:00000004 r6:00000001 r5:01af1040
[  139.400000] r4:00000004
[  144.400000] PM: suspend of devices complete after 5010.000 msecs
[  144.400000] PM: late suspend of devices complete after 0.001 msecs
[  144.400000] PM: noirq suspend of devices complete after 0.001 msecs
[  144.400000] clk source suspend
[  144.400000] clk event suspend
[  144.400000] clk event resume
[  144.400000] clk source resume
[  144.400000] 
[  144.400000]  KPI irq status: 0x5 
[  144.400000] PM: noirq resume of devices complete after 0.001 msecs
[  144.400000] PM: early resume of devices complete after 0.001 msecs
[  144.400000] nuc970fb resume
[  144.400000] Warning, request gpio fail...
[  144.490000] PM: resume of devices complete after 90.000 msecs
[  144.820000] Restarting tasks ... done.

junwushi, 2017-06-12:
i think this code has issue:
static int nuc970fb_suspend(struct platform_device *dev, pm_message_t state)
{
	struct fb_info	   *fbinfo = platform_get_drvdata(dev);
	struct nuc970fb_info *info = fbinfo->par;
	unsigned long timeout;
		
	printk(KERN_INFO "nuc970fb suspend\n");
	info->powerdown = 1;
	
	timeout = **_wait_for_completion_interruptible_timeout_**
			 (&info->completion, NUC970_FB_TIMEOUT);
			
	nuc970fb_stop_lcd(fbinfo);

	return 0;
}

junwushi, 2017-06-12:
and same issue in ehci-hcd.c at the  function ehci_suspend that used msleep function.

int ehci_suspend(struct usb_hcd *hcd, bool do_wakeup)
{
	struct ehci_hcd		*ehci = hcd_to_ehci(hcd);

	if (time_before(jiffies, ehci->next_statechange))
		msleep(10);

	/*
	 * Root hub was already suspended.  Disable IRQ emission and
	 * mark HW unaccessible.  The PM and USB cores make sure that
	 * the root hub is either suspended or stopped.
	 */
	ehci_prepare_ports_for_controller_suspend(ehci, do_wakeup);

	spin_lock_irq(&ehci->lock);
	ehci_writel(ehci, 0, &ehci->regs->intr_enable);
	(void) ehci_readl(ehci, &ehci->regs->intr_enable);

	clear_bit(HCD_FLAG_HW_ACCESSIBLE, &hcd->flags);
	spin_unlock_irq(&ehci->lock);

	return 0;
}

mjchen1, 2017-06-13:
I think, maybe config not correct.
could you test following flow:
1. make nuc972_defconfig
[image]
2. selection kpi releation setting in menuconfig
[image]
3. make kernel again
4. retry kpi wake-up: 
    echo mem > /sys/power/state
     then press kpi key

In my test, I can not find problem.
my test board is "NHS-972-1-CY-1M53".

junwushi, 2017-06-13:
now wakeup is not problem,thanks. but that issues are show "BUG: scheduling while atomic", please google it why occurrenced. many driver's suspend use the like function "msleep()" in suspend function ,that is not allow. got it? my English is not good.

yachen, 2017-06-13:
Hi @junwushi,

We'll check if the msleep() in suspend is a bug or not.
The msleep() in ehci_suspend() still exist in 4.x kernel, and seems no one is complain about it.

中文測試...

Sincerely,

Yi-An Chen

closed 2017-06-30 by junwushi