OpenNuvoton/NUC970_Linux_Kernel #7: after suspend_resume the lcd can't display old image

junwushi, 2017-06-30:
lcd is mpu80 type 240X320 size,control chip is st7789v ,it is like ili9431.
when enter suspend state,if kernel wakeup  by key pressed,but the display is blank, LCD  keep VDD supply and reset pin always keep high when suspend. i hasn't idea for it. pls help me.
[nuc900fb.c.tar.gz](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/1114952/nuc900fb.c.tar.gz)
root@HHT:~# echo mem >/sys/power/state 
[  142.290000] PM: Syncing filesystems ... done.
[  142.300000] Freezing user space processes ... (elapsed 0.02 seconds) done.
[  142.320000] Freezing remaining freezable tasks ... (elapsed 0.02 seconds) done.
[  142.340000] Suspending console(s) (use no_console_suspend to debug)
[  142.350000] nuc970fb suspend
[  146.290000]  KPI key: 0x17 
[  147.210000]  KPI key: 0x13 
[  147.350000] PM: suspend of devices complete after 5010.000 msecs
[  147.350000] PM: late suspend of devices complete after 0.001 msecs
[  147.350000] PM: noirq suspend of devices complete after 0.001 msecs
[  147.350000] clk source suspend
[  147.350000] clk event suspend
[  147.350000] clk event resume
[  147.350000] clk source resume
[  147.350000] PM: noirq resume of devices complete after 0.001 msecs
[  147.350000] PM: early resume of devices complete after 0.001 msecs
[  147.350000] nuc970fb resume
[  147.350000] Warning, request gpio fail...
[  147.350000] Warning, request gpio fail...
[  147.350000] Warning, request gpio fail...
[  147.440000] PM: resume of devices complete after 90.000 msecs
[  147.490000]  KPI key: 0x17 
[  147.510000] Restarting tasks ... done.

yachen, 2017-07-03:
You attached a NUC900 frame buffer driver?

junwushi, 2017-07-04:
yes

yachen, 2017-07-04:
Why you did attach nuc900 frame buffer driver for nuc970 issue?

junwushi, 2017-07-04:
sorry, I  upload error file.
[nuc970fb.c.tar.gz](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/1121504/nuc970fb.c.tar.gz)

yachen, 2017-07-05:
Hi, 

I don't have a board with MPU panel. But I built the kernel image with MPU panel driver enabled for testing.  There's a bug in frame buffer driver but you seems fixed it by calling init_ST7789V() after frame buffer driver resumed. I also check LCM register and LCD interface waveform, all seems working properly. So I don't have any clue about what goes wrong yet.

Can you check the waveform of LCD interface after system resume? and read back the registers of ST7789V to make sure the setting is correct. Thanks.

Sincerely,

Yi-An Chen

junwushi, 2017-07-07:
init_ST7789V() is reset and reinit all register for LCD,it is need more time. that one way but not better to sloved it

yachen, 2017-07-10:
Hi,

So you mean the driver you sent me already fixed this issue, except it takes too long to re-init the panel?

Sincerely,

Yi-An Chen

junwushi, 2017-07-11:
no, it is only driver st7789 chip,and after resume,the LCD working, if not reinit st7789, the LCD can't show draw any . i am in user application first save screen and then  restore screen after resume . now is my solution.

yachen, 2017-07-11:
Hi,

Try replace init_ST7789V() function call with following line in nuc970fb_resume() and see if the display can display the last frame before enter power dowm mode or not.
writel(readl(regs + REG_LCM_DCCS) | LCM_DCCS_VA_EN, regs + REG_LCM_DCCS);

Sincerely,

Yi-An Chen

junwushi, 2017-07-11:
yes,seems working,but image not all restore,only gettty image restore,the application image lost,why?
see two

[image]

 image file.

junwushi, 2017-07-11:
it is new drvier file. that reduce resume time only. and solved LCM working after resume. the image restore still need application to do it.

[nuc970fb.c.tar.gz](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/1138046/nuc970fb.c.tar.gz)

yachen, 2017-07-11:
The NUC970 frame buffer driver doesn't not wipe out the frame buffer memory. 
Probably virtual terminal triggers a repaint  after resume.

closed 2017-07-14 by junwushi