OpenNuvoton/NUC970_Linux_Kernel #8: ttyGS0 after wakeup not working

junwushi, 2017-07-03:
please check.

yachen, 2017-07-04:
What USB to serial device are you using? FTDI, Prolific, or others?

junwushi, 2017-07-04:
gatget

yachen, 2017-07-04:
Hi, yes I know it's the gadget device. May I know its controller? We've tested other gadget device and function perperly after systrem wake up. I'd like to test the same device if possible.

junwushi, 2017-07-04:
there not cable with controller?only usb cable connect pc in usb0

yachen, 2017-07-04:
Are you testing USB host or USB device of NUC970?

junwushi, 2017-07-04:
yes

junwushi, 2017-07-04:
U-Boot 2013.04-rc2-ge877f9c-dirty (May 29 2017 - 15:08:53)

CPU: NUC972
DRAM:  64 MiB
NAND:  128 MiB
MMC:   mmc: 0
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   emac
Hit any key to stop autoboot:  0 

Loading from nand0, offset 0x200000
   Image Name:   Linux-3.10.106-00955-g6ec51e32a2
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3050528 Bytes = 2.9 MiB
   Load Address: 00007fc0
   Entry Point:  00008000
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-3.10.106-00955-g6ec51e32a2
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3050528 Bytes = 2.9 MiB
   Load Address: 00007fc0
   Entry Point:  00008000
   Verifying Checksum ... [    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.106-00955-g6ec51e32a27-dirty (jwshi@leo-VirtualBox) (gcc version 6.2.0 (GCC) ) #123 PREEMPT Mon Jul 3 18:21:12 CST 2017
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: NUC970
[    0.000000] cma: CMA: reserved 4 MiB at 03c00000
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rw rootflags=inband-tags console=ttyS0,115200n8 init=/sbin/init mem=64M lpj=744448 vt.global_cursor_default=0
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 64MB = 64MB total
[    0.000000] Memory: 54364k/54364k available, 11172k reserved, 0K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     vmalloc : 0xc4800000 - 0xff000000   ( 936 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc059eecc   (5724 kB)
[    0.000000]       .init : 0xc059f000 - 0xc05d6240   ( 221 kB)
[    0.000000]       .data : 0xc05d8000 - 0xc0622980   ( 299 kB)
[    0.000000]        .bss : 0xc0622980 - 0xc064bd6c   ( 165 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:625
[    0.000000] sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 4294967286ms
[    0.000000] Console: colour dummy device 80x30
[    0.000000] console [ttyS0] enabled
[    0.130000] Calibrating delay loop (skipped) preset value.. 148.88 BogoMIPS (lpj=744448)
[    0.140000] pid_max: default: 32768 minimum: 301
[    0.150000] Mount-cache hash table entries: 512
[    0.150000] Initializing cgroup subsys devices
[    0.160000] Initializing cgroup subsys freezer
[    0.160000] Initializing cgroup subsys perf_event
[    0.170000] CPU: Testing write buffer coherency: ok
[    0.170000] Setting up static identity map for 0xc04715a0 - 0xc04715f8
[    0.180000] devtmpfs: initialized
[    0.270000] pinctrl core: initialized pinctrl subsystem
[    0.280000] NET: Registered protocol family 16
[... 136 lines cut ...]
[    2.110000] lib80211: common routines for IEEE802.11 drivers
[    2.110000] NET: Registered protocol family 36
[    2.130000] Key type encrypted registered
[    2.130000] g_serial gadget: high-speed config #2: CDC ACM config
[    2.140000] nuc970-rtc nuc970-rtc: setting system clock to 2017-02-09 03:55:13 UTC (1486612513)
[    2.150000] yaffs: dev is 32505858 name is "mtdblock2" rw
[    2.160000] yaffs: passed flags "inband-tags"
[    3.540000] VFS: Mounted root (yaffs2 filesystem) on device 31:2.
[    3.540000] devtmpfs: mounted
[    3.550000] Freeing unused kernel memory: 220K (c059f000 - c05d6000)
INIT~version 2.88 boot
Starting udev
[    4.370000] udevd[681]: starting version 3.2
[    4.500000] udevd[682]: starting eudev-3.2
Populating dev cache
INIT: Entering runlevel: 5
Configuring network interfaces... done.
Starting syslogd/klogd: done
nohup: appending output to nohup.out
[   10.310000] input: virtual keyboard as /devices/virtual/input/input1

HHT /dev/ttyS0

HHT login: root
root@HHT:~#

junwushi, 2017-07-04:
wait i am check in EV board

junwushi, 2017-07-04:
inittab file content

# /etc/inittab: init(8) configuration.
# $Id: inittab,v 1.91 2002/01/25 13:35:21 miquels Exp $

# The default runlevel.
id:5:initdefault:

# Boot-time system configuration/initialization script.
# This is run first except when booting in emergency (-b) mode.
si::sysinit:/etc/init.d/rcS

# What to do in single-user mode.
~~:S:wait:/sbin/sulogin

# /etc/init.d executes the S and K scripts upon change
# of runlevel.
#
# Runlevel 0 is halt.
# Runlevel 1 is single-user.
# Runlevels 2-5 are multi-user.
# Runlevel 6 is reboot.

l0:0:wait:/etc/init.d/rc 0
l1:1:wait:/etc/init.d/rc 1
l2:2:wait:/etc/init.d/rc 2
l3:3:wait:/etc/init.d/rc 3
l4:4:wait:/etc/init.d/rc 4
l5:5:wait:/etc/init.d/rc 5
l6:6:wait:/etc/init.d/rc 6
# Normally not reached, but fallthrough in case of emergency.
z6:6:respawn:/sbin/sulogin

# /sbin/getty invocations for the runlevels.
#
# The "id" field MUST be the same as the last
# characters of the device (after "tty").
#
# Format:
#  <id>:<runlevels>:<action>:<process>
#

0:123456:respawn:/sbin/getty 115200 ttyS0
1:123456:respawn:/sbin/getty 115200 ttyGS0

yachen, 2017-07-04:
I don't understand why did you post an inittab file here. This is not the file we included in our BSP.

junwushi, 2017-07-04:
1:123456:respawn:/sbin/getty 115200 ttyGS0   
that means how to use the ttyGS0 for testing ok or not

closed 2017-07-14 by junwushi