OpenNuvoton/NUC970_Linux_Kernel #9: can add feature for gpio-poweroff if kenal without device tree option

junwushi, 2017-07-10:
the file ./drivers/power/reset/gpio-poweroff.c i check that is only support device tree kennel for define power off GPIO pin,but i am not use device tree to build kernel, i want define it in dev.c. can add this feature?

junwushi, 2017-07-11:
other question,please help me check.i want read wake source register value in user application. but always read value is zero. why?

#define SYS_BA 0xb0000000
#define REG_WKUPSSR 0x5c
unsigned int getwakupsource(void) {
	unsigned int value;
	int mem_fd = open("/dev/mem", O_RDWR | O_SYNC);
	if (mem_fd == -1) {
		printf("open /dev/mem fail\n");
		return 0;
	}
	unsigned char *virt_regaddr = (unsigned char*) mmap(0, 0x1000,
			PROT_READ | PROT_WRITE, MAP_SHARED, mem_fd, SYS_BA);
	if (virt_regaddr == NULL) {
		close(mem_fd);
		printf("mmap mem fail\n");
		return 0;
	}
	value = *((unsigned int*) (virt_regaddr));
	printf("PID:0x%x\n", value);
	value = *((unsigned int *) (virt_regaddr + REG_WKUPSSR));
	munmap(virt_regaddr, 0x1000);
	close(mem_fd);
	return value;
}

yachen, 2017-07-11:
Hi,

Because the wake up source is cleared in arch/arm/mach-nuc970/pm.c after resume. Please csearch following line in pm.c
__raw_writel(__raw_readl(REG_WKUPSER), REG_WKUPSSR);	// clear wake source flag

Sincerely,

Yi-An Chen

junwushi, 2017-07-11:
i think that put it at fisrt line is better in the funcation of pm.c?

yachen, 2017-07-11:
Yup, updated. Relocate the clear wakeup flag code.

junwushi, 2017-07-11:
please check i give address is right in above function for device /dev/mem. it is phy address?,because i  still can not get right status value of wake source register. when key wakeup i get 0x800000(134217728) and rtc alarm wakeup get 0x1997a0(1677216). what's wrong hapended?

yachen, 2017-07-11:
Following code works fine here. Not sure what goes wrong at your environment.
```
#include <stdio.h>
#include <stdlib.h>
#include <unistd.h>
#include <fcntl.h>
#include <string.h>
#include <sys/mman.h>

#define DEV 	"/dev/mem"
#define BA	0xB0000000
int main(int argc, char **argv)
{
	int offset, fd;
	unsigned char *va;

	if(argc == 1) {
		printf("Usage: mem <offset>\n");
		return 0;
	}

	offset = atol(argv[1]);
	
	fd = open(DEV, O_RDWR | O_SYNC);
	if(fd < 0) {
		printf("Open /dev/mem failed\n");
		return 0;
	}

	va = mmap(0, 0x1000, PROT_READ | PROT_WRITE, MAP_SHARED, fd, BA);
	if(va == NULL) {
		printf("mmap failed\n");
		return 0;
	}

	printf("=>%08x\n", *(unsigned int volatile *)(va + offset));

	munmap(va, 0x1000);
	close(fd);
}
```

yachen, 2017-07-12:
If you simply want to use GPIO to detect power off event, you can put it in NUC970 GPIO driver.

junwushi, 2017-07-12:
no,i want after system shutdown to poweroff main power of device.so need override pm_power_0ff() funtion.it is below
/*
 * hhu-poweroff.c
 *
 *  Created on: 2017年7月10日
 *      Author: jwshi
 */

/*
 * Toggles a GPIO pin to power down a device
 *
 * Jamie Lentin <jm@lentin.co.uk>
 * Andrew Lunn <andrew@lunn.ch>
 *
 * Copyright (C) 2012 Jamie Lentin
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License version 2 as
 * published by the Free Software Foundation.
 *
 */
#include <linux/kernel.h>
#include <linux/init.h>
#include <linux/delay.h>
#include <linux/platform_device.h>
#include <linux/gpio.h>
#include <linux/of_platform.h>
#include <linux/of_gpio.h>
#include <linux/module.h>

/*
 * Hold configuration here, cannot be more than one instance of the driver
 * since pm_power_off itself is global.
 */
static int gpio_num = -1;
//static int gpio_active_low;

static void gpio_poweroff_do_poweroff(void) {
	//BUG_ON(!gpio_is_valid(gpio_num));

	/* drive it active, also inactive->active edge */
	//gpio_direction_output(gpio_num, !gpio_active_low);
	//mdelay(100);
	/* drive inactive, also active->inactive edge */
	//gpio_set_value(gpio_num, gpio_active_low);
	//mdelay(100);
	/* drive it active, also inactive->active edge */
	gpio_direction_output(gpio_num, 1);
	//printk("power down now\n");
	/* give it some time */
//	mdelay(3000);
	//WARN_ON(1);
}

static int gpio_poweroff_probe(struct platform_device *pdev) {
//	enum of_gpio_flags flags;
	//bool input = false;
	int ret;
	//printk("%s:probe1\n",__func__);
	/* If a pm_power_off function has already been added, leave it alone */
	if (pm_power_off != NULL) {
		pr_err("%s: pm_power_off function already registered", __func__);
		return -EBUSY;
	}
	gpio_num = NUC970_PE4;
	//gpio_num = of_get_gpio_flags(pdev->dev.of_node, 0, &flags);
	if (!gpio_is_valid(gpio_num))
	{
		pr_err("%s: gpio num is not valid", __func__);
		return gpio_num;
	}
	//gpio_active_low = flags & OF_GPIO_ACTIVE_LOW;

	//input = of_property_read_bool(pdev->dev.of_node, "input");

	ret = gpio_request(gpio_num, NULL);
	if (ret) {
		pr_err("%s: Could not get GPIO %d", __func__, gpio_num);
		return ret;
	}

	if (gpio_direction_output(gpio_num, 0)) {
		pr_err("Could not set direction of GPIO %d", gpio_num);
		goto err;
	}

	pm_power_off = &gpio_poweroff_do_poweroff;
	//printk("%s:probe2\n",__func__);
	return 0;

	err: gpio_free(gpio_num);
	return -ENODEV;
}

static int gpio_poweroff_remove(struct platform_device *pdev) {
	gpio_free(gpio_num);
	if (pm_power_off == &gpio_poweroff_do_poweroff)
		pm_power_off = NULL;

	return 0;
}

#if defined(CONFIG_OF)
static const struct of_device_id of_gpio_poweroff_match[] = {
	{   .compatible = "poweroff-gpio"},
	{  },
};
MODULE_DEVICE_TABLE(of, of_gpio_poweroff_match);
#endif

static struct platform_driver gpio_poweroff_driver = {
	.probe = gpio_poweroff_probe,
	.remove = gpio_poweroff_remove,
	.driver = {
		.name = "poweroff-gpio",
		.owner = THIS_MODULE,
#if defined(CONFIG_OF)
		.of_match_table = of_match_ptr(of_gpio_poweroff_match),
#endif
	},
};

module_platform_driver(gpio_poweroff_driver);

MODULE_DESCRIPTION("GPIO poweroff driver");
MODULE_LICENSE("GPL v2");

closed 2017-07-14 by junwushi