OpenNuvoton/NUC970_Linux_Kernel #10: open irda port report  msb_index() bug

junwushi, 2017-07-14:
i am use uart5 connect IRDA transceiver chip. after open("/dev/ttyS5") and  setbaud() etc. parameter final use ioctl(irda_fd, TIOCSETD, &irdadisc);
 irdadisc is  N_IRDA.
but  the kernal report below message:

[  107.190000] msb_index(), Detected buggy peer, adjust null PV to 0x1!

what 's happend?

junwushi, 2017-07-14:
i use your bsp provide dems name irda_demos ,it is show same issue,pls check soon.
thanks.

yachen, 2017-07-14:
Did you configure kernel as below?
[*] Networking support ---> 
    <*> IrDA (infrared) subsystem support ---> 
     Infrared-port device drivers ---> 
             <*> NUC970 SIR on UART

junwushi, 2017-07-14:
yes,configure it already.

yachen, 2017-08-14:
Hi,

Please pull commit f862b85 and test again.

Sincerely,

Yi-An Chen

junwushi, 2017-08-15:
yes, when close irda it is not crash now.
but are you test irda comunication working or not . i am still can't comunication with irda.

yachen, 2017-08-15:
Yes we did and it works fine.

junwushi, 2017-08-15:
but i am only can send data and receive failed,and i am check rxd the signal is arrived to nuc972 uart5 rx pin.

yachen, 2017-08-15:
The IrDA SIR protocol is half-duplex only, was the TX active while RX receiving data?

junwushi, 2017-08-17:
when open Irda port and after resume that crashed.

[  346.100000] nuc970fb suspend
[  346.120000] PM: suspend of devices complete after 30.000 msecs
[  346.120000] PM: late suspend of devices complete after 0.001 msecs
[  346.120000] PM: noirq suspend of devices complete after 0.001 msecs
[  346.120000] clk source suspend
[  346.120000] clk event suspend
[  346.120000] clk event resume
[  346.120000] clk source resume
[  346.120000] PM: noirq resume of devices complete after 0.001 msecs
[  346.120000] PM: early resume of devices complete after 0.001 msecs
[  346.120000] nuc970fb resume
[  346.210000] PM: resume of devices complete after 90.000 msecs
[  346.260000] Restarting tasks ... done.
[  346.530000] Unable to handle kernel NULL pointer dereference at virtual address 0000005c
[  346.530000] pgd = c3764000
[  346.530000] [0000005c] *pgd=036f3831, *pte=00000000, *ppte=00000000
[  346.530000] Internal error: Oops: 17 [#1] PREEMPT ARM
[  346.530000] Modules linked in:
[  346.530000] CPU: 0 PID: 793 Comm: example Not tainted 3.10.107-01233-g4f949220bd0-dirty #187
[  346.530000] task: c34ba000 ti: c36bc000 task.ti: c36bc000
[  346.530000] PC is at nuc970serial_stop_tx+0x30/0x74
[  346.530000] LR is at uart_stop+0x44/0x68
[  346.530000] pc : [<c021bf00>]    lr : [<c0217c34>]    psr: 60000093
[  346.530000] sp : c36bddf8  ip : c36bde08  fp : c36bde04
[  346.530000] r10: 00000005  r9 : c36bc000  r8 : 00000000
[  346.530000] r7 : c3571618  r6 : c34bff00  r5 : a0000013  r4 : c36bc000
[  346.530000] r3 : f8000500  r2 : 00000000  r1 : 00000000  r0 : c05ca984
[  346.530000] Flags: nZCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  346.530000] Control: 0005317f  Table: 03764000  DAC: 00000015
[  346.530000] Process example (pid: 793, stack limit = 0xc36bc1b8)
[  346.530000] Stack: (0xc36bddf8 to 0xc36be000)
[  346.530000] dde0:                                                       c36bde1c c36bde08
[  346.530000] de00: c0217c34 c021bee0 c3571600 c379b0a0 c36bde2c c36bde20 c02615f8 c0217c00
[  346.530000] de20: c36bde44 c36bde30 c02054e4 c02615e0 c3571600 c3571600 c36bde6c c36bde48
[  346.530000] de40: c0205840 c02054a0 00000000 c3571600 c34bff00 c056e008 00000000 c36bc000
[  346.530000] de60: c36bde84 c36bde70 c020690c c0205828 c3571600 00000000 c36bdf14 c36bde88
[  346.530000] de80: c01fefd0 c02068ec c0046f98 00000000 c35716a8 00000000 00000120 00000001
[  346.530000] dea0: c36bdee4 c36bc000 60000013 00011357 00000007 00000007 c34bff00 c36bc000
[  346.530000] dec0: c342a460 00000007 c36bdf44 00000001 c34bff08 00000008 c3571600 00000000
[  346.530000] dee0: 00000000 e71637d3 c3693cf0 c34bff00 c300ca18 c342a470 00000000 c3693cf0
[  346.530000] df00: c3693cf0 00000008 c36bdf54 c36bdf18 c00c8c58 c01fec34 00000000 00000000
[  346.530000] df20: 00000007 c34bff08 c36bdf4c 00000000 c34ba000 00000000 c05bd4e8 c36bdfb0
[  346.530000] df40: c36bc000 00000000 c36bdf64 c36bdf58 c00c8e80 c00c8bd4 c36bdf84 c36bdf68
[  346.530000] df60: c003ae60 c00c8e80 c36bc000 c36bc000 c36bc000 c000f0a8 c36bdfac c36bdf88
[  346.530000] df80: c0011f10 c003adc8 b6f9183c 00011334 0000008d 00000006 c000f0a8 c36bc000
[  346.530000] dfa0: 00000000 c36bdfb0 c000ef44 c0011e7c 00000000 00000002 b6f97124 00000000
[  346.530000] dfc0: b6f9183c 00011334 0000008d 00000006 00000000 000215dc beba5a14 00000000
[  346.530000] dfe0: 00000000 beba59f8 b6f5eb34 b6f5f3a4 80000010 00000005 3f551000 b7ce1000
[  346.530000] Backtrace: 
[  346.530000] [<c021bed0>] (nuc970serial_stop_tx+0x0/0x74) from [<c0217c34>] (uart_stop+0x44/0x68)
[  346.530000] [<c0217bf0>] (uart_stop+0x0/0x68) from [<c02615f8>] (nuc970irda_close+0x28/0x2c)
[  346.530000]  r5:c379b0a0 r4:c3571600
[  346.530000] [<c02615d0>] (nuc970irda_close+0x0/0x2c) from [<c02054e4>] (tty_ldisc_close+0x54/0x6c)
[  346.530000] [<c0205490>] (tty_ldisc_close+0x0/0x6c) from [<c0205840>] (tty_ldisc_kill+0x28/0x138)
[  346.530000]  r5:c3571600 r4:c3571600
[  346.530000] [<c0205818>] (tty_ldisc_kill+0x0/0x138) from [<c020690c>] (tty_ldisc_release+0x30/0x50)
[  346.530000]  r9:c36bc000 r8:00000000 r7:c056e008 r6:c34bff00 r5:c3571600
r4:00000000
[  346.530000] [<c02068dc>] (tty_ldisc_release+0x0/0x50) from [<c01fefd0>] (tty_release+0x3ac/0x654)
[  346.530000]  r5:00000000 r4:c3571600
[  346.530000] [<c01fec24>] (tty_release+0x0/0x654) from [<c00c8c58>] (__fput+0x94/0x274)
[  346.530000] [<c00c8bc4>] (__fput+0x0/0x274) from [<c00c8e80>] (____fput+0x10/0x14)
[  346.530000] [<c00c8e70>] (____fput+0x0/0x14) from [<c003ae60>] (task_work_run+0xa8/0xbc)
[  346.530000] [<c003adb8>] (task_work_run+0x0/0xbc) from [<c0011f10>] (do_work_pending+0xa4/0xd0)
[  346.530000]  r7:c000f0a8 r6:c36bc000 r5:c36bc000 r4:c36bc000
[  346.530000] [<c0011e6c>] (do_work_pending+0x0/0xd0) from [<c000ef44>] (work_pending+0xc/0x20)
[  346.530000]  r9:c36bc000 r8:c000f0a8 r7:00000006 r6:0000008d r5:00011334
r4:b6f9183c
[  346.530000] Code: 0a000002 e3c22002 e5903004 e5832004 (e5d1305c) 
[  346.530000] ---[ end trace f1f6b12b86125618 ]---
[  346.540000] note: example[793] exited with preempt_count 1

junwushi, 2017-11-13:
i an still can not receive any char .but send char feature is fine,not found reason.

junwushi, 2017-11-22:
can share the irda parts schematic?

yachen, 2017-11-23:
[image]

yachen, 2017-11-23:
Hi @junwushi , 

Please let me know if you also need the OrCAD file.

Sincerely,

Yi-An Chen

junwushi, 2017-11-23:
don't need or-cad file,I am use sensor is TFDU4101, it is different hsdl-3201?

yachen, 2017-11-24:
We haven't used TFDU4101 before.

yachen, 2018-01-18:
Close unless there's further discussion on this issue.

closed 2018-01-18 by yachen