OpenNuvoton/NUC970_Linux_Kernel #11: Problem of Configuring Timers Subsystem

morway, 2017-07-18:
When I use 'make menuconfig' to configure kernel, I can't find 'High Resolution Timer Support' while previous versions had it. The path is as follows:
->General setup 
    ->Timers subsystem
       ->[ ]High Resolution Timer Support
How could I find this line? Thanks!

yachen, 2017-07-18:
Hi,

Timer will operates in one shot mode after high resolution timer enabled. And NUC970 timer has a limitation in one shot mode that it cannot be restart right after it's stopped. So in some rare case, system can stop forever.
It's disabled in commit a8d1402.  You can revert this commit if you really needs high resolution timer. Or use ETIMER instead, ETIMER doesn't have this limitation.

Sincerely,

Yi-An Chen

morway, 2017-07-19:
Okay. I will try it. Thanks!

closed 2017-07-20 by yachen