OpenNuvoton/NUC970_Linux_Kernel #12: nuc970 eth0 driver bug

xuganlai, 2017-08-14:
now, i do a experience:continuous uninterrupted on/off machine.occasionally the system print error below:
-----------------------------------------------------------------------------------------------
nuc970-emac0 nuc970-emac0: eth0 is OPENED
skb->tail:c2a4001f,skb->end:c2a30620,len:65535--------(add by owner)recevied data len 65535???
skbuff: skb_over_panic: text:c02483f0 len:65535 put:65535 head:c2a18000 data:c2a18020 tail:0xc2a2801f end:0xc2a18620 dev:eth0
------------[ cut here ]------------
kernel BUG at net/core/skbuff.c:96!
Internal error: Oops - BUG: 0 [#1] PREEMPT ARM
Modules linked in:
CPU: 0 PID: 0 Comm: swapper Not tainted 3.10.32 #27
task: c0457b18 ti: c044e000 task.ti: c044e000
PC is at skb_put+0x98/0xac
LR is at console_unlock+0x340/0x530
pc : [<c02be060>]    lr : [<c001f804>]    psr: 60000013
sp : c044fdc8  ip : c044fcf0  fp : c044fe04
r10: 00000000  r9 : c280ba20  r8 : c2a18620
r7 : 0000ffff  r6 : 0000ffff  r5 : c2a2801f  r4 : c2a18020
r3 : 60000013  r2 : 60000013  r1 : c0488e18  r0 : 0000007d
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
Control: 0005317f  Table: 02a48000  DAC: 00000017
Process swapper (pid: 0, stack limit = 0xc044e1b8)
Stack: (0xc044fdc8 to 0xc0450000)
fdc0:                   0000ffff c2a18000 c2a18020 c2a2801f c2a18620 c280b800
fde0: c02c14bc 0000ffff c28b9e40 c280bbc0 c2a20600 c304a000 c044fe4c c044fe08
fe00: c02483f0 c02bdfd8 00000000 c044fe18 c01fae9c 00000010 c280bc4c c280b800
fe20: ffff8bf7 00000010 c280bc4c 00000003 0000012c c0487a40 c0487a48 ffff8bf9
fe40: c044fe7c c044fe50 c02c7ca0 c0248348 00000000 0000000c 00000001 00000003
fe60: c04a93ec 00000101 0000000a c044e000 c044febc c044fe80 c0025a30 c02c7bdc
fe80: c045f5f4 c292ad80 ffff8bf8 00200000 41069265 60000093 c04736ec 00000000
fea0: c044ff54 00000001 41069265 c0456000 c044fed4 c044fec0 c0025be8 c0025984
fec0: c0072638 00000013 c044feec c044fed8 c0025c50 c0025ba4 c006f428 00000013
fee0: c044ff0c c044fef0 c00105c0 c0025c04 41069265 c0010e3c 60000013 f0000000
ff00: c044ff1c c044ff10 c0008410 c0010590 c044ff74 c044ff20 c000f330 c0008410
ff20: 00000000 0005317f 0005217f 60000013 c044e000 c0456078 c0487b05 c0487b05
ff40: 00000001 41069265 c0456000 c044ff74 600000d3 c044ff68 c0010e30 c0010e3c
ff60: 60000013 ffffffff c044ff9c c044ff78 c004f7c8 c0010e0c c044ff9c 00000002
ff80: c0487bcc c0487bc0 c044997c ffffffff c044ffb4 c044ffa0 c035242c c004f718
ffa0: c0487bcc c04560f0 c044fff4 c044ffb8 c042cbbc c03523c4 ffffffff ffffffff
ffc0: c042c734 00000000 00000000 c044997c 00053175 c045601c c0449d80 c045855c
ffe0: 00004000 00448abc 00000000 c044fff8 00008040 c042c98c 00000000 00000000
Backtrace: 
[<c02bdfc8>] (skb_put+0x0/0xac) from [<c02483f0>] (nuc970_poll+0xb8/0x2b4)
 r8:c304a000 r7:c2a20600 r6:c280bbc0 r5:c28b9e40 r4:0000ffff
[<c0248338>] (nuc970_poll+0x0/0x2b4) from [<c02c7ca0>] (net_rx_action+0xd4/0x1d0)
[<c02c7bcc>] (net_rx_action+0x0/0x1d0) from [<c0025a30>] (__do_softirq+0xbc/0x1bc)
[<c0025974>] (__do_softirq+0x0/0x1bc) from [<c0025be8>] (do_softirq+0x54/0x60)
[<c0025b94>] (do_softirq+0x0/0x60) from [<c0025c50>] (irq_exit+0x5c/0x9c)
 r4:00000013
[<c0025bf4>] (irq_exit+0x0/0x9c) from [<c00105c0>] (handle_IRQ+0x40/0x90)
 r4:00000013
[<c0010580>] (handle_IRQ+0x0/0x90) from [<c0008410>] (asm_do_IRQ+0x10/0x14)
 r6:f0000000 r5:60000013 r4:c0010e3c
[<c0008400>] (asm_do_IRQ+0x0/0x14) from [<c000f330>] (__irq_svc+0x30/0x74)
Exception stack(0xc044ff20 to 0xc044ff68)
ff20: 00000000 0005317f 0005217f 60000013 c044e000 c0456078 c0487b05 c0487b05
ff40: 00000001 41069265 c0456000 c044ff74 600000d3 c044ff68 c0010e30 c0010e3c
ff60: 60000013 ffffffff
[<c0010dfc>] (arch_cpu_idle+0x0/0x48) from [<c004f7c8>] (cpu_startup_entry+0xc0/0xfc)
[<c004f708>] (cpu_startup_entry+0x0/0xfc) from [<c035242c>] (rest_init+0x78/0x90)
 r8:ffffffff r7:c044997c r6:c0487bc0 r5:c0487bcc r4:00000002
[<c03523b4>] (rest_init+0x0/0x90) from [<c042cbbc>] (start_kernel+0x240/0x2d8)
 r4:c04560f0
[<c042c97c>] (start_kernel+0x0/0x2d8) from [<00008040>] (0x8040)
Code: e58d500c e58d8010 e58dc014 eb025317 (e7f001f2) 
---[ end trace 383f2c805f6bce4f ]---
Kernel panic - not syncing: Fatal exception in interrupt

yachen, 2017-08-14:
Hi,

Have you pulled latest EMAC driver?
Now packet length larger than 1514 (not counting 4-byte CRC) will not pass to socket layer.

Sincerely,

Yi-An Chen

xuganlai, 2017-08-15:
thanks for a reply, i have replaced with the latest kernel version.now it is testing.
unfortunately,have found new question.972 demo board  ping other machine,display below:
64 bytes from 192.168.1.5 :seq=0 ttl=128 time=0.000ms   <----------time show 0.000ms?????

yachen, 2017-08-15:
Hi,

Due to a timer issue. High resolution timer and no HZ support is disabled for NUC970 in a8d1402.
So Kernel cannot tell time difference smaller than 10ms.

Sincerely,

Yi-An Chen

xuganlai, 2017-08-15:
that is to say can not be resolved?is it hardware problem?
high resolution timer and no HZ support  is disabled,so which system functions will be affected?
thanks.

yachen, 2017-08-15:
It's a hardware limitation. If you really needs HR timer or no HZ, use ETIMER as system timer.

Sincerely,

Yi-An Chen

xuganlai, 2017-08-15:
how to modify ETIMER as system timer.
thanks.

yachen, 2017-08-16:
Please replace the TIMER register controls in arch/arm/mach-nuc972/time.c with ETIMER registers. And also update relative clock/interrupt settings.

Sincerely,

Yi-An Chen

closed 2017-08-22 by yachen