OpenNuvoton/NUC970_Linux_Kernel #15: usb gsm modems drviers error

junwushi, 2017-08-29:
usb1 connect  UC20 modem module,and show below error info:
i search  the reason on internet, they are said usb time setting error.

[    2.690000] usb 2-2: new low-speed USB device number 2 using nuc970-ohci
[    2.880000] usb 2-2: device descriptor read/64, error -62
[    3.170000] usb 2-2: device descriptor read/64, error -62
[    3.460000] usb 2-2: new low-speed USB device number 3 using nuc970-ohci
[    3.650000] usb 2-2: device descriptor read/64, error -62

junwushi, 2017-08-29:
please check soon

yachen, 2017-08-29:
Nope, this is not a clock setting issue.
From you log, the modem module was identified as a low speed device, which it is not. 
So it's more likely to be D+/D- signal issue. What board are you using?

Sincerely,

Yi-An Chen

junwushi, 2017-08-29:
i am in my self device,the modem is usb 2.0 device not low speed device, and not give advise for pull-up resistor in dp or dp pin in the modem hardware design document

yachen, 2017-08-29:
Hi,

Below is the USB layout guideline in our document.

1. DP/DM traces should be length matched and as close as possible to the connector.
2. Route DP/DM traces should be close together for noise rejection on the differential
signals, parallel to each other and the length difference within 200-mil.
3. If the common chock is necessary, it should be as close as to the connector.
4. No extra components at DP/DM pair traces to maintain signal integrity.
5. No de-coupled caps on the DP/DM.
6. The characteristic of matching impedance 90 Ohm on the DP/DM is necessary.
7. Stubs on high speed USB signals should be avoided, as stubs will cause signal
reflections and affect signal quality. If a stub is unavoidable in the design, no stub
should be greater than 200 mils.
8. Route all traces over continuous planes (VCC or GND), with no interruptions.
Avoid crossing over anti-etch if at all possible. Crossing over anti-etch (plane
splits) increases inductance and radiation levels by forcing a greater loop area.
Likewise, avoid changing layers with high-speed traces as much as practical. It is
preferable to change layers to avoid crossing a plane split.
9. Use the following guidelines for the VCC OR GND plane.
a. Traces should not cross anti-etch, for it greatly increases the return path for
those signal traces. This applies to High Speed USB signals, high-speed clock
and signal traces as well as slower signal traces, which might be coupling to
them. USB signaling is not purely differential in all speeds (i.e. the FS Single
Ended Zero is common mode)
b. Avoid routing of USB signals within 25 mils of any anti-etch to avoid coupling to
the next split or radiating from the edge of the PCB.
10. Separate signal traces into similar categories and route similar signal traces
together (such as routing differential pairs together).
11. Keep high-speed USB signals clear of the core logic set. High current transients
are produced during internal state transitions and can be very difficult to filter out.
12. Follow the 20*h thumb rule by keeping traces at least 20*(height above the plane)
away from the edge of the plane (VCC or GND, depending on the plane the trace
is over). For the suggested stack up the height above the plane are 4.5 mils. This
calculates to a 90-mil spacing requirement from the edge of the plane. This helps
prevent the coupling of the signal onto adjacent wires and also helps prevent free
radiation of the signal from the edge of the PCB.

Sincerely,

Yi-An Chen

junwushi, 2017-08-29:
sorry maybe harware problem,i am change another borad ,it is working.

closed 2017-08-29 by junwushi