OpenNuvoton/NUC970_Linux_Kernel #17: why define two same scancode KEY_RESERVED in diffreent key in nuc970_keypad.c,how to indentify it?

junwushi, 2017-10-27:
static int nuc970_keymap[] = {
	// row 0
	KEY_RESERVED,	KEY_LEFTALT, 	KEY_F5,         KEY_INFO,
	KEY_UP,	        KEY_F7,         KEY_TAB,	KEY_F1,

	// row 1
	KEY_ESC,	KEY_DOWN,	KEY_RIGHT,	KEY_F6,
	KEY_F2,	        KEY_1,          KEY_2,	        KEY_4,

	// row 2
	KEY_LEFT,	KEY_CAPSLOCK,   KEY_F3,          KEY_B,
	KEY_5,          KEY_6,          KEY_SPACE,       KEY_F4,

	// row 3
	KEY_RESERVED,	KEY_7,          KEY_8,	         KEY_9,
	KEY_BACKSPACE,	KEY_DOT,        KEY_0,	         KEY_ENTER,
};

mjchen1, 2017-11-13:
The keymap struct just show user how to define this struct.  
usually, user will re-define this keymap.

junwushi, 2017-11-14:
i known,but different key map is better,now I am not change the code from git hub. I am only use it and switch to myself key map in my application like input method.

closed 2017-11-14 by junwushi