OpenNuvoton/NUC970_Linux_Kernel #19: after update 2c90956      Add power-on-setting check to nand driver  then can't mount the rootfs

junwushi, 2017-11-14:
1.780000] DCCP: Activated CCID 2 (TCP-like)
[    1.790000] DCCP: Activated CCID 3 (TCP-Friendly Rate Control)
[    1.800000] sctp: Hash tables configured (established 2048 bind 4096)
[    1.810000] NET: Registered protocol family 21
[    1.820000] Registered RDS/tcp transport
[    1.820000] NET: Registered protocol family 36
[    1.830000] Key type dns_resolver registered
[    1.840000] Key type encrypted registered
[    1.850000] nuc970-rtc nuc970-rtc: setting system clock to 2017-02-14 09:34:07 UTC (1487064847)
[    1.860000] yaffs: dev is 32505858 name is "mtdblock2" rw
[    1.860000] yaffs: passed flags "inband-tags"
[    2.020000] VFS: Mounted root (yaffs2 filesystem) on device 31:2.
[    2.020000] devtmpfs: error mounting -2
[    2.030000] Freeing unused kernel memory: 160K
[    2.040000] Failed to execute /sbin/init.  Attempting defaults...
[    2.040000] Kernel panic - not syncing: No init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidance.
[    2.040000] CPU: 0 PID: 1 Comm: swapper Not tainted 3.10.107-01388-g2c90956bcb1-dirty #15
[    2.040000] Backtrace: 
[    2.040000] [<c00122ec>] (dump_backtrace+0x0/0x108) from [<c00124fc>] (show_stack+0x18/0x1c)
[    2.040000]  r7:00000000 r6:00000000 r5:c05aa8c0 r4:c05aaec8
[    2.040000] [<c00124e4>] (show_stack+0x0/0x1c) from [<c042585c>] (dump_stack+0x20/0x28)
[    2.040000] [<c042583c>] (dump_stack+0x0/0x28) from [<c0424d0c>] (panic+0xa0/0x1f8)
[    2.040000] [<c0424c6c>] (panic+0x0/0x1f8) from [<c042460c>] (kernel_init+0x128/0x154)
[    2.040000]  r3:c065a070 r2:00000527 r1:c3603100 r0:c04d723c
[    2.040000]  r7:00000000
[    2.040000] [<c04244e4>] (kernel_init+0x0/0x154) from [<c000ef98>] (ret_from_fork+0x14/0x3c)
[    2.040000]  r5:c04244e4 r4:00000000

junwushi, 2017-11-14:
U-Boot 2013.04-rc2-g8a5cd0c-dirty (Aug 29 2017 - 11:26:58)

CPU: NUC972
DRAM:  64 MiB
NAND:  128 MiB
MMC:   mmc: 0
*** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   emac
Hit any key to stop autoboot:  0 

Loading from nand0, offset 0x200000
   Image Name:   Linux-3.10.107-01388-g2c90956bcb
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2858992 Bytes = 2.7 MiB
   Load Address: 00007fc0
   Entry Point:  00008000
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-3.10.107-01388-g2c90956bcb
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2858992 Bytes = 2.7 MiB
   Load Address: 00007fc0
   Entry Point:  00008000
   Verifying Checksum ...     0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.107-01388-g2c90956bcb1-dirty (jwshi@leo-VirtualBox) (gcc version 6.2.0 (GCC) ) #15 PREEMPT Tue Nov 14 11:43:01 CST 2017
[    0.000000] CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
[    0.000000] CPU: VIVT data cache, VIVT instruction cache
[    0.000000] Machine: NUC970
[    0.000000] cma: CMA: reserved 4 MiB at 03c00000
[    0.000000] Memory policy: ECC disabled, Data cache writeback
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: noinitrd root=/dev/mtdblock2 rootfstype=yaffs2 rw rootflags=inband-tags console=ttyS0,115200n8 init=/sbin/init mem=64M lpj=744448 vt.global_cursor_default=0
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Memory: 64MB = 64MB total
[    0.000000] Memory: 54828k/54828k available, 10708k reserved, 0K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     vmalloc : 0xc4800000 - 0xff000000   ( 936 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
[    0.000000]     modules : 0xbf000000 - 0xc0000000   (  16 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0537ec4   (5312 kB)
[    0.000000]       .init : 0xc0538000 - 0xc0560004   ( 161 kB)
[    0.000000]       .data : 0xc0562000 - 0xc05aa8c0   ( 291 kB)
[    0.000000]        .bss : 0xc05aa8c0 - 0xc05d7a7c   ( 181 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] NR_IRQS:625
[    0.000000] sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 4294967286ms
[    0.000000] Console: colour dummy device 80x30
[    0.000000] console [ttyS0] enabled
[    0.130000] Calibrating delay loop (skipped) preset value.. 148.88 BogoMIPS (lpj=744448)
[    0.140000] pid_max: default: 32768 minimum: 301
[    0.150000] Mount-cache hash table entries: 512
[    0.150000] Initializing cgroup subsys devices
[    0.160000] Initializing cgroup subsys freezer
[    0.160000] Initializing cgroup subsys perf_event
[    0.170000] CPU: Testing write buffer coherency: ok
[    0.170000] Setting up static identity map for 0xc04288e8 - 0xc0428924
[    0.180000] devtmpfs: initialized
[    0.270000] pinctrl core: initialized pinctrl subsystem
[    0.280000] NET: Registered protocol family 16
[... 125 lines cut ...]
[    1.800000] sctp: Hash tables configured (established 2048 bind 4096)
[    1.810000] NET: Registered protocol family 21
[    1.820000] Registered RDS/tcp transport
[    1.820000] NET: Registered protocol family 36
[    1.830000] Key type dns_resolver registered
[    1.840000] Key type encrypted registered
[    1.850000] nuc970-rtc nuc970-rtc: setting system clock to 2017-02-14 09:34:07 UTC (1487064847)
[    1.860000] yaffs: dev is 32505858 name is "mtdblock2" rw
[    1.860000] yaffs: passed flags "inband-tags"
[    2.020000] VFS: Mounted root (yaffs2 filesystem) on device 31:2.
[    2.020000] devtmpfs: error mounting -2
[    2.030000] Freeing unused kernel memory: 160K
[    2.040000] Failed to execute /sbin/init.  Attempting defaults...
[    2.040000] Kernel panic - not syncing: No init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidance.
[    2.040000] CPU: 0 PID: 1 Comm: swapper Not tainted 3.10.107-01388-g2c90956bcb1-dirty #15
[    2.040000] Backtrace: 
[    2.040000] [<c00122ec>] (dump_backtrace+0x0/0x108) from [<c00124fc>] (show_stack+0x18/0x1c)
[    2.040000]  r7:00000000 r6:00000000 r5:c05aa8c0 r4:c05aaec8
[    2.040000] [<c00124e4>] (show_stack+0x0/0x1c) from [<c042585c>] (dump_stack+0x20/0x28)
[    2.040000] [<c042583c>] (dump_stack+0x0/0x28) from [<c0424d0c>] (panic+0xa0/0x1f8)
[    2.040000] [<c0424c6c>] (panic+0x0/0x1f8) from [<c042460c>] (kernel_init+0x128/0x154)
[    2.040000]  r3:c065a070 r2:00000527 r1:c3603100 r0:c04d723c
[    2.040000]  r7:00000000
[    2.040000] [<c04244e4>] (kernel_init+0x0/0x154) from [<c000ef98>] (ret_from_fork+0x14/0x3c)
[    2.040000]  r5:c04244e4 r4:00000000

yachen, 2017-11-14:
Hi,

We're unable to reproduce this problem. 
Did you program the yaffs2 partition to NAND flash by U-Boot or NuWriter?
If you restore the kernel image (with previous NAND driver), it can mount this MTP partition without problem?

Sincerely,

Yi-An Chen

yachen, 2017-11-14:
Hi,

Did you set power on setting about ECC and page size?

Sincerely,

Yi-An Chen

junwushi, 2017-11-14:
cfg9 and cfg8 dip are all off? how does setting it?

junwushi, 2017-11-14:
i am setting PA6-P9 all off ,it is working, others failed.

yachen, 2017-11-14:
Hi,

For the Winbond NAND 128MiB 3,3V 8-bit flash you used, please set PA6~PA9 as OFF.
In path 2c90956,  we updated NAND driver to use power on setting to overwrite the flash type found by kernel. So since this flash can be identified by kernel alright, you should set PA6~PA9 as OFF.

Sincerely,

Yi-An Chen

closed 2017-11-15 by junwushi