OpenNuvoton/NUC970_Linux_Kernel #24: nuc975 eth0 get skb buffer error

xuganlai, 2018-03-29:
kernel:3.10.107.
board:nuc975.
web server app runs on the board.
the browser transfer files to the board via post method.the kernel print below(the web server is performing well on nuc951 board-RAM is 32M):
------------------------------------------
BC: page allocation failure: order:0, mode:0x20	
CPU: 0 PID: 552 Comm: BC Tainted: G           O 3.10.107 #26
Backtrace: 
[<c0012d78>] (dump_backtrace+0x0/0x108) from [<c0012e98>] (show_stack+0x18/0x1c)
 r6:c0da0000 r5:00000000 r4:00000020
[<c0012e80>] (show_stack+0x0/0x1c) from [<c0358964>] (dump_stack+0x20/0x2c)
[<c0358944>] (dump_stack+0x0/0x2c) from [<c008b5c0>] (warn_alloc_failed+0xcc/0x124)
[<c008b4f4>] (warn_alloc_failed+0x0/0x124) from [<c008c04c>] (__alloc_pages_nodemask+0x694/0x910)
 r3:00000000 r2:00000000
 r7:00000020 r6:c0da0000 r5:00000000 r4:c0492ffc
[<c008b9b8>] (__alloc_pages_nodemask+0x0/0x910) from [<c02c6600>] (__netdev_alloc_frag+0xa4/0x15c)
[<c02c655c>] (__netdev_alloc_frag+0x0/0x15c) from [<c02c6748>] (__netdev_alloc_skb+0x90/0x100)
[<c02c66b8>] (__netdev_alloc_skb+0x0/0x100) from [<c024d198>] (nuc970_poll+0x1f0/0x2c4)
 r6:c080bbc0 r5:0000003c r4:c0f7d780
[<c024cfa8>] (nuc970_poll+0x0/0x2c4) from [<c02cd144>] (net_rx_action+0xd4/0x1d0)
[<c02cd070>] (net_rx_action+0x0/0x1d0) from [<c0026378>] (__do_softirq+0xbc/0x1bc)
[<c00262bc>] (__do_softirq+0x0/0x1bc) from [<c002655c>] (do_softirq+0x54/0x60)
[<c0026508>] (do_softirq+0x0/0x60) from [<c00265c4>] (irq_exit+0x5c/0x9c)
 r4:00000013
[<c0026568>] (irq_exit+0x0/0x9c) from [<c00105e0>] (handle_IRQ+0x40/0x90)
 r4:00000013
[<c00105a0>] (handle_IRQ+0x0/0x90) from [<c0008410>] (asm_do_IRQ+0x10/0x14)
 r6:f0000000 r5:60000013 r4:c0234e0c
[<c0008400>] (asm_do_IRQ+0x0/0x14) from [<c000f330>] (__irq_svc+0x30/0x74)
Exception stack(0xc0da17b0 to 0xc0da17f8)
17a0:                                     c080b010 f000d000 c0da0000 f000d000
17c0: 00380400 00000000 c080b020 00000000 00000000 00000040 c080b010 c0da180c
17e0: c0da1810 c0da17f8 c0235090 c0234e0c 60000013 ffffffff
[<c0234de0>] (nuc970_check_rb+0x0/0x54) from [<c0235090>] (nuc970_nand_command_lp+0x214/0x3f8)
 r4:00000000
[<c0234e7c>] (nuc970_nand_command_lp+0x0/0x3f8) from [<c02354ec>] (nuc970_nand_read_page_hwecc_oob_first+0x48/0x250)
[<c02354a4>] (nuc970_nand_read_page_hwecc_oob_first+0x0/0x250) from [<c022f3d4>] (nand_do_read_ops+0x17c/0x46c)
[<c022f258>] (nand_do_read_ops+0x0/0x46c) from [<c022f744>] (nand_read_oob+0x80/0x3b4)
[<c022f6c4>] (nand_read_oob+0x0/0x3b4) from [<c0227678>] (part_read_oob+0x84/0x168)
[<c02275f4>] (part_read_oob+0x0/0x168) from [<c0223a40>] (mtd_read_oob+0x3c/0x6c)
[<c0223a04>] (mtd_read_oob+0x0/0x6c) from [<c0174290>] (yaffs_mtd_read+0x80/0x124)
 r4:00000800
[<c0174210>] (yaffs_mtd_read+0x0/0x124) from [<c0173a44>] (yaffs_tags_marshall_read+0x164/0x1f4)
[<c01738e0>] (yaffs_tags_marshall_read+0x0/0x1f4) from [<c0172f18>] (yaffs_rd_chunk_tags_nand+0x44/0xcc)
[<c0172ed4>] (yaffs_rd_chunk_tags_nand+0x0/0xcc) from [<c016ca9c>] (yaffs_rd_data_obj+0x68/0x80)
 r7:c0cfe000 r6:c0cfe000 r5:c0726800 r4:c0d524a8
[<c016ca34>] (yaffs_rd_data_obj+0x0/0x80) from [<c016fd80>] (yaffs_file_rd+0x28c/0x294)
 r5:c0d524a8 r4:00000070
[<c016faf4>] (yaffs_file_rd+0x0/0x294) from [<c0167b4c>] (yaffs_readpage_nolock+0x98/0x188)
[<c0167ab4>] (yaffs_readpage_nolock+0x0/0x188) from [<c0167df0>] (yaffs_readpage+0x34/0x78)
[<c0167dbc>] (yaffs_readpage+0x0/0x78) from [<c008ee7c>] (__do_page_cache_readahead+0x254/0x2c0)
 r6:0000001f r5:00000008 r4:c04d54c0
[<c008ec28>] (__do_page_cache_readahead+0x0/0x2c0) from [<c008ef14>] (ra_submit+0x2c/0x34)
[<c008eee8>] (ra_submit+0x0/0x34) from [<c00845cc>] (filemap_fault+0x408/0x450)
[<c00841c4>] (filemap_fault+0x0/0x450) from [<c00a2408>] (__do_fault+0x78/0x4a8)
[<c00a2390>] (__do_fault+0x0/0x4a8) from [<c00a33f8>] (handle_pte_fault+0x7c/0x730)
[<c00a337c>] (handle_pte_fault+0x0/0x730) from [<c00a42a0>] (handle_mm_fault+0xbc/0xd4)
[<c00a41e4>] (handle_mm_fault+0x0/0xd4) from [<c001631c>] (do_page_fault+0x238/0x3a4)
[<c00160e4>] (do_page_fault+0x0/0x3a4) from [<c0016598>] (do_translation_fault+0x84/0xb4)
[<c0016514>] (do_translation_fault+0x0/0xb4) from [<c0008614>] (do_PrefetchAbort+0x40/0xa4)
 r8:c0da1fb0 r7:c046094c r6:00000005 r5:20000010 r4:c04608fc
[<c00085d4>] (do_PrefetchAbort+0x0/0xa4) from [<c000f698>] (ret_from_exception+0x0/0x10)
Exception stack(0xc0da1fb0 to 0xc0da1ff8)
1fa0:                                     000027f1 00000bb8 000027f1 00000bb8
1fc0: b6c8ddbc be5ffe20 b6c8e004 00000000 00000000 00000000 be5ffe20 be5ff90c
1fe0: be5ff8ff be5ff0e8 0001bdb8 000477f0 20000010 ffffffff
Mem-info:
Normal per-cpu:
CPU    0: hi:    0, btch:   1 usd:   0
active_anon:173 inactive_anon:1 isolated_anon:0
[... 68 lines cut ...]
[<c0016514>] (do_translation_fault+0x0/0xb4) from [<c0008614>] (do_PrefetchAbort+0x40/0xa4)
 r8:c0da1fb0 r7:c046094c r6:00000005 r5:20000010 r4:c04608fc
[<c00085d4>] (do_PrefetchAbort+0x0/0xa4) from [<c000f698>] (ret_from_exception+0x0/0x10)
Exception stack(0xc0da1fb0 to 0xc0da1ff8)
1fa0:                                     000027f1 00000bb8 000027f1 00000bb8
1fc0: b6c8ddbc be5ffe20 b6c8e004 00000000 00000000 00000000 be5ffe20 be5ff90c
1fe0: be5ff8ff be5ff0e8 0001bdb8 000477f0 20000010 ffffffff
Mem-info:
Normal per-cpu:
CPU    0: hi:    0, btch:   1 usd:   0
active_anon:173 inactive_anon:1 isolated_anon:0
 active_file:0 inactive_file:168 isolated_file:0
 unevictable:1471 dirty:0 writeback:0 unstable:0
 free:4049 slab_reclaimable:198 slab_unreclaimable:422
 mapped:51 shmem:2 pagetables:31 bounce:0
 free_cma:4018
Normal free:16196kB min:420kB low:524kB high:628kB active_anon:692kB inactive_anon:4kB active_file:0kB inactive_file:672kB unevictable:5884kB isolated(anon):0kB isolated(file):0kB present:32768kB managed:11120kB mlocked:0kB dirty:0kB writeback:0kB mapped:204kB shmem:8kB slab_reclaimable:792kB slab_unreclaimable:1688kB kernel_stack:296kB pagetables:124kB unstable:0kB bounce:0kB free_cma:16072kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? no
lowmem_reserve[]: 0 0
Normal: 5*4kB (UC) 4*8kB (UMC) 5*16kB (MC) 2*32kB (MC) 2*64kB (C) 2*128kB (C) 1*256kB (C) 2*512kB (C) 2*1024kB (C) 2*2048kB (C) 2*4096kB (C) = 16196kB
1646 total pagecache pages
0 pages in swap cache
Swap cache stats: add 0, delete 0, find 0/0
Free swap  = 0kB
Total swap = 0kB
nuc970-emac0 nuc970-emac0: get skb buffer error

yachen, 2018-03-29:
Hi,

Can you provide your kernel config?
Also enabled kernel memory leak debugger to check if there' s memory leak. Thanks.

Sincerely,

Yi-An Chen

xuganlai, 2018-03-29:
attachment is kernel config.
how to config to enable kernel memory debugger?
thanks

xuganlai, 2018-03-29:
how to enable kernel memory leak debugger?
could you provide your email address?
I don't know why I can't add attachments here.
thanks

yachen, 2018-03-29:
Hi,

Enable  CONFIG_DEBUG_KMEMLEAK, rebuild kernel. And execute in shell.
mount -t debugfs nodev /sys/kernel/debug/
cat /sys/kernel/debug/kmemleak

You can paste your config as plan text here in this thread.

Sincerely,

Yi-An Chen

xuganlai, 2018-03-29:
#
# Automatically generated file; DO NOT EDIT.
# Linux/arm 3.10.107 Kernel Configuration
#
CONFIG_ARM=y
CONFIG_SYS_SUPPORTS_APM_EMULATION=y
CONFIG_HAVE_PROC_CPU=y
CONFIG_STACKTRACE_SUPPORT=y
CONFIG_HAVE_LATENCYTOP_SUPPORT=y
CONFIG_LOCKDEP_SUPPORT=y
CONFIG_TRACE_IRQFLAGS_SUPPORT=y
CONFIG_RWSEM_GENERIC_SPINLOCK=y
CONFIG_GENERIC_HWEIGHT=y
CONFIG_GENERIC_CALIBRATE_DELAY=y
CONFIG_NEED_DMA_MAP_STATE=y
CONFIG_FIQ=y
CONFIG_VECTORS_BASE=0xffff0000
CONFIG_ARM_PATCH_PHYS_VIRT=y
CONFIG_NEED_MACH_GPIO_H=y
CONFIG_GENERIC_BUG=y
CONFIG_DEFCONFIG_LIST="/lib/modules/$UNAME_RELEASE/.config"
CONFIG_IRQ_WORK=y
CONFIG_BUILDTIME_EXTABLE_SORT=y

#
# General setup
#
CONFIG_BROKEN_ON_SMP=y
CONFIG_INIT_ENV_ARG_LIMIT=32
CONFIG_CROSS_COMPILE=""
CONFIG_LOCALVERSION=""
# CONFIG_LOCALVERSION_AUTO is not set
CONFIG_HAVE_KERNEL_GZIP=y
CONFIG_HAVE_KERNEL_LZMA=y
CONFIG_HAVE_KERNEL_XZ=y
CONFIG_HAVE_KERNEL_LZO=y
CONFIG_KERNEL_GZIP=y
# CONFIG_KERNEL_LZMA is not set
# CONFIG_KERNEL_XZ is not set
# CONFIG_KERNEL_LZO is not set
CONFIG_DEFAULT_HOSTNAME="(none)"
CONFIG_SWAP=y
CONFIG_SYSVIPC=y
CONFIG_SYSVIPC_SYSCTL=y
# CONFIG_POSIX_MQUEUE is not set
CONFIG_FHANDLE=y
# CONFIG_AUDIT is not set
CONFIG_HAVE_GENERIC_HARDIRQS=y

#
# IRQ subsystem
#
CONFIG_GENERIC_HARDIRQS=y
CONFIG_GENERIC_IRQ_PROBE=y
CONFIG_GENERIC_IRQ_SHOW=y
CONFIG_HARDIRQS_SW_RESEND=y
CONFIG_KTIME_SCALAR=y
CONFIG_ARCH_USES_GETTIMEOFFSET=y
CONFIG_GENERIC_CLOCKEVENTS=y
CONFIG_GENERIC_CLOCKEVENTS_BUILD=y

#
# Timers subsystem
#
CONFIG_HZ_PERIODIC=y

#
# CPU/Task time and stats accounting
#
CONFIG_TICK_CPU_ACCOUNTING=y
[... 1784 lines cut ...]
# CONFIG_CRC8 is not set
CONFIG_ZLIB_INFLATE=y
CONFIG_ZLIB_DEFLATE=y
CONFIG_LZO_COMPRESS=y
CONFIG_LZO_DECOMPRESS=y
CONFIG_XZ_DEC=y
# CONFIG_XZ_DEC_X86 is not set
# CONFIG_XZ_DEC_POWERPC is not set
# CONFIG_XZ_DEC_IA64 is not set
# CONFIG_XZ_DEC_ARM is not set
# CONFIG_XZ_DEC_ARMTHUMB is not set
# CONFIG_XZ_DEC_SPARC is not set
# CONFIG_XZ_DEC_BCJ is not set
# CONFIG_XZ_DEC_TEST is not set
CONFIG_HAS_IOMEM=y
CONFIG_HAS_IOPORT=y
CONFIG_HAS_DMA=y
CONFIG_DQL=y
CONFIG_NLATTR=y
CONFIG_GENERIC_ATOMIC64=y
CONFIG_ARCH_HAS_ATOMIC64_DEC_IF_POSITIVE=y
# CONFIG_AVERAGE is not set
# CONFIG_CORDIC is not set
# CONFIG_DDR is not set
# CONFIG_VIRTUALIZATION is not set

xuganlai, 2018-03-29:
1、have configed: (1)kernel debugging (2) kernel memory leak detector
2、recompiled and downloaded
3、but kernel print:
kmemleak:kernel memory leak detector disabled
kmemleak:early log buffer exceeded (2034),please increase DEBUG_KMEMLEAK_EARLY_LOG_SIZE

so last cannot found kmemleak in /sys/kernel/debug.
which steps are ignored;
thanks

yachen, 2018-04-11:
Hi, 

As the log message shows, please try to increase DEBUG_KMEMLEAK_EARLY_LOG_SIZE and test again. Thanks.

Sincerely,

Yi-An Chen

xuganlai, 2018-04-11:
now ubi have been replaced by yaffs2.but system up have a issue,which
steps is ignored? (kernel Related options of yaffs2 are selected.and make rootfs:mkyaffs2 --inband-tags -p 2048 rootfs rootfs_yaffs2.img .)Attach print message:

cma: CMA: reserved 16 MiB at 01000000
Memory policy: ECC disabled, Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
Kernel command line: noinitrd root=/dev/mtdblock2 rw rootfstype=yaffs2 console=ttyS0,115200n8 rdinit=/sbin/init mem=32M mtdparts=nand0:0x100000@0x0(u-boot),0x500000@0x100000(kernel),0x2000000@0x600000(rootfs),-(user) mac=00:00:00:11:66:88
PID hash table entries: 128 (order: -3, 512 bytes)
Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
Inode-cache hash table entries: 2048 (order: 1, 8192 bytes)
Memory: 32MB = 32MB total
Memory: 11328k/11328k available, 21440k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    vmalloc : 0xc2800000 - 0xff000000   ( 968 MB)
    lowmem  : 0xc0000000 - 0xc2000000   (  32 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc03d5fd4   (3896 kB)
      .init : 0xc03d6000 - 0xc0420968   ( 299 kB)
      .data : 0xc0422000 - 0xc045e960   ( 243 kB)
       .bss : 0xc045e960 - 0xc0492e40   ( 210 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Preemptible hierarchical RCU implementation.
NR_IRQS:625
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 4294967286ms
Console: colour dummy device 80x30
console [ttyS0] enabled
kmemleak: Kernel memory leak detector disabled
kmemleak: Early log buffer exceeded (2034), please increase DEBUG_KMEMLEAK_EARLY_LOG_SIZE
Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 512
Initializing cgroup subsys devices
Initializing cgroup subsys freezer
Initializing cgroup subsys perf_event
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0xc030e378 - 0xc030e3b4
devtmpfs: initialized
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
bio: create slab <bio-0> at 0
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
i2c_gpio_probe - pdev = i2c-gpio
platform i2c-gpio.2: Driver i2c-gpio requests probe deferral
Advanced Linux Sound Architecture Driver Initialized.
NET: Registered protocol family 2
TCP established hash table entries: 512 (order: 0, 4096 bytes)
TCP bind hash table entries: 512 (order: -1, 2048 bytes)
TCP: Hash tables configured (established 512 bind 512)
TCP: reno registered
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
NetWinder Floating Point Emulator V0.97 (double precision)
msgmni has been set to 54
NET: Registered protocol family 38
Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254)
io scheduler noop registered
io scheduler deadline registered
io scheduler cfq registered (default)
nuc970_dma_probe - pdev = nuc970-dma-m2m
nuc970-dma nuc970-dma-m2m: NUC970 M2M DMA ready
nuc970-uart.0: ttyS0 at I/O 0x0 (irq = 36) is a NUC970
nuc970-uart.10: ttyS10 at I/O 0x0 (irq = 42) is a NUC970
brd: module loaded
[... 36 lines cut ...]
platform nuc970-audio: Driver nuc970-audio requests probe deferral
TCP: cubic registered
NET: Registered protocol family 17
NET: Registered protocol family 15
i2c_gpio_probe - pdev = i2c-gpio
i2c-gpio i2c-gpio.2: using pins 193 (SDA) and 192 (SCL)
nuc970-audio nuc970-audio:  nau8822-hifi <-> nuc970-audio-i2s mapping ok
ALSA device list:
  #0: nuc970_IIS
yaffs: dev is 32505858 name is "mtdblock2" rw
yaffs: passed flags ""
VFS: Mounted root (yaffs2 filesystem) on device 31:2.
devtmpfs: error mounting -2
Freeing unused kernel memory: 296K (c03d6000 - c0420000)
Kernel panic - not syncing: No init found.  Try passing init= option to kernel. See Linux Documentation/init.txt for guidance.
CPU: 0 PID: 1 Comm: swapper Not tainted 3.10.107 #32
Backtrace: 
[<c0012f08>] (dump_backtrace+0x0/0x108) from [<c0013028>] (show_stack+0x18/0x1c)
 r6:00000000 r5:c042b914 r4:c042b914
[<c0013010>] (show_stack+0x0/0x1c) from [<c030b994>] (dump_stack+0x20/0x2c)
[<c030b974>] (dump_stack+0x0/0x2c) from [<c030b630>] (panic+0x84/0x204)
[<c030b5ac>] (panic+0x0/0x204) from [<c030ab0c>] (kernel_init+0x134/0x164)
 r3:000003a0 r2:c04d5070 r1:c04d5070 r0:c03869ac
[<c030a9d8>] (kernel_init+0x0/0x164) from [<c000f7b8>] (ret_from_fork+0x14/0x3c)
 r5:c030a9d8 r4:00000000

xuganlai, 2018-04-15:
modify as follows.but there's still the same issue.
noinitrd root=/dev/mtdblock2 rw rootfstype=yaffs2 rootflags=inband-tags console=ttyS0,115200n8 rdinit=/sbin/init mem=32M mtdparts=nand0:0x100000@0x0(u-boot),0x500000@0x100000(kernel),0x2000000@0x600000(rootfs),-(user)

yachen, 2018-04-17:
Hi, Please try to use initrd as rootfs and mount yaffs2 partition manually. If it failed to mount manually, probably the yaffs2 partition is not write into NAND correctly.

yachen, 2018-08-30:
Close unless there's new feedback.

closed 2018-08-30 by yachen