OpenNuvoton/NUC970_Linux_Kernel #28: USB EHCI Debug Port support in kernel and hardware

ponyatov, 2018-09-19:
https://www.coreboot.org/EHCI_Debug_Port

Are any attempts were done to enable EHCI-interfaced hardware debugging in nuc970 SoCs?

ponyatov, 2018-09-19:
The advantage compared to classical JTAG is there is no requirement to have specially designed PCB with JTAG connector (I have no one module with JTAG on my desk now). You just plug USB cable to the device (yes, you lost and lock it only for debug session), and plug another end to a generic USB port on PC, no Jlink, no extra hw, works off the shelf (if enabled in).

yachen, 2018-09-20:
Hi,

Our EHCI does not support EHCI appendix C. So it cannot support EHCI debug port.

Sincerely.

Yi-An Chen

closed 2018-09-20 by yachen