OpenNuvoton/NUC970_Linux_Kernel #31: In Dual mode reading from SPI flash gives wrong data

lightelf76, 2019-02-18:
I am bringing up custom board with NUC976 CPU. SPI NOR flash (W25Q128 or GD25Q127 with 4 wires) on SPI0 used as main storage. UBoot and Kernel loads OK, but attempt to mount rootfs (squashfs or jffs2) fails with multiple CRC errors. 
I was tried to comment out a M25P80_DUAL_READ flag in m25p80.c for my flash chip and errors disappear.
So: SPI0 works nice in Normal SPI mode, but gives wrong data in attempt to use Dual SPI mode.

cwweng-nuvoton, 2019-02-19:
Hi,
I cannot  reproduce your issue.
Did you modify any source code ?
Please provide your kernel config for us to check this issue.

yachen, 2019-03-27:
Please reopen and provide kernel config if this issue still exists

closed 2019-03-27 by yachen

lightelf76, 2019-03-27:
Yes, it seems related to my kernel CONFIG/changes, I need to bissect to find issues. Clean build (TOMATO_DEFCONFIG) from OpenNuvoton kernel sources works as expected. Will check later.