OpenNuvoton/NUC970_Linux_Kernel #37: NUC972 network issue

qianfan-Zhao, 2019-04-24:
My device can't connnect the host by using tcp protocol sometimes after reboot. I don't know the reason now.

The device has a static ip address after power on: 193.168.18.20/24, there has another win10 computer(193.168.18.3/24) connected it. 

Nexit is the network informations:
```
eth0      Link encap:Ethernet  HWaddr 00:DF:C1:A8:12:14
          inet addr:193.168.18.20  Bcast:193.168.18.255  Mask:255.255.255.0
          UP BROADCAST RUNNING  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:54 errors:0 dropped:0 overruns:0 frame:0
          TX packets:54 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:5164 (5.0 KiB)  TX bytes:5164 (5.0 KiB)

# route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         193.168.18.1    0.0.0.0         UG    0      0        0 eth0
193.168.18.0    0.0.0.0         255.255.255.0   U     0      0        0 eth0
```

I can't ping any other ip address other than localhost.

```
# tcpdump -i lo & 
# ping 193.168.18.3
PING 193.168.18.3 (193.168.18.3): 56 data bytes
^C
--- 193.168.18.3 ping statistics ---
3 packets transmitted, 0 packets received, 100% packet loss
# 17:47:39.360000 IP 193.168.18.20 > 193.168.18.20: ICMP host 193.168.18.3 unreachable, length 92
17:47:39.360000 IP 193.168.18.20 > 193.168.18.20: ICMP host 193.168.18.3 unreachable, length 92
17:47:39.360000 IP 193.168.18.20 > 193.168.18.20: ICMP host 193.168.18.3 unreachable, length 92

#
# ping 193.168.18.20
PING 193.168.18.20 (193.168.18.20): 56 data bytes
17:47:56.200000 IP 193.168.18.20 > 193.168.18.20: ICMP echo request, id 11778, seq 0, length 64
17:47:56.200000 IP 193.168.18.20 > 193.168.18.20: ICMP echo reply, id 11778, seq 0, length 64
64 bytes from 193.168.18.20: seq=0 ttl=64 time=0.000 ms
17:47:57.210000 IP 193.168.18.20 > 193.168.18.20: ICMP echo request, id 11778, seq 1, length 64
17:47:57.210000 IP 193.168.18.20 > 193.168.18.20: ICMP echo reply, id 11778, seq 1, length 64
64 bytes from 193.168.18.20: seq=1 ttl=64 time=0.000 ms
17:47:58.220000 IP 193.168.18.20 > 193.168.18.20: ICMP echo request, id 11778, seq 2, length 64
17:47:58.220000 IP 193.168.18.20 > 193.168.18.20: ICMP echo reply, id 11778, seq 2, length 64
64 bytes from 193.168.18.20: seq=2 ttl=64 time=0.000 ms
17:47:59.230000 IP 193.168.18.20 > 193.168.18.20: ICMP echo request, id 11778, seq 3, length 64
17:47:59.230000 IP 193.168.18.20 > 193.168.18.20: ICMP echo reply, id 11778, seq 3, length 64
64 bytes from 193.168.18.20: seq=3 ttl=64 time=0.000 ms
^C
--- 193.168.18.20 ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 0.000/0.000/0.000 ms
#
```

I don't know why all the packets can't send from eth0, and who cause this problem, emac driver or tcp stack or another?

yachen, 2019-04-24:
Tx/Rx packet counts are both 0. What hardware are you testing with? Which PHY chip are you using?
Can you check "cat /proc/interrupts" and see if there's any EMAC interrupt?
Also please check if there's anything abnormal about the REFCLK signal.

Sincerely,

Yi-An Chen

qianfan-Zhao, 2019-04-24:
The problem doesn't come up every time, about 10 percent of the time. I think it is ok with our board or PHY chip.

I added printk information to the nuc970_ether_start_xmit function. When the problem occurs, you can see that the Linux network protocol stack does call the nuc970_ether_start_xmit function, but no packets are sent from the hardware interface.

Here is the interrupt message:
```
# cat /proc/interrupts
           CPU0
 16:     384842         -  nuc970-timer0
 18:         28         -  nuc970adc
 19:          0         -  b0002000.emac0
 21:          0         -  b0002000.emac0
 23:          0         -  ehci_hcd:usb1
 24:          0         -  ohci_hcd:usb2
 36:        438         -  ttyS0
 38:         12         -
 45:        152         -
 52:       4110         -  b8006300.spi1
Err:          0
```

qianfan-Zhao, 2019-04-24:
**nuc970_reset_mac(dev, 0);** can't resume if this problem happens.

```
# ifconfig eth0 down; ifconfig eth0 up
nuc970_ether_stats
nuc970-emac0 b0002000.emac0: eth0 is OPENED
nuc970_ether_stats
# nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit
nuc970_ether_start_xmit 2

# cat /proc/interrupts
           CPU0
 16:      42545         -  nuc970-timer0
 18:        413         -  nuc970adc
 19:          0         -  b0002000.emac0
 21:          0         -  b0002000.emac0
 23:          0         -  ehci_hcd:usb1
 24:          0         -  ohci_hcd:usb2
 36:        505         -  ttyS0
 37:          0         -  ttyS1
 38:         48         -  ttyS2
 39:          0         -  ttyS4
 41:          0         -  ttyS8
 45:       1219         -  ttyS7
 52:      55973         -  b8006300.spi1
Err:          0
#
```

```
diff --git a/drivers/net/ethernet/nuvoton/nuc970_ether0.c b/drivers/net/ethernet/nuvoton/nuc970_ether0.c
index d53f04bb7b7..e2f4267db7d 100755
--- a/drivers/net/ethernet/nuvoton/nuc970_ether0.c
+++ b/drivers/net/ethernet/nuvoton/nuc970_ether0.c
@@ -633,6 +633,7 @@ static int nuc970_ether_close(struct net_device *dev)
 static struct net_device_stats *nuc970_ether_stats(struct net_device *dev)
 {
        struct nuc970_ether *ether;
+       printk("%s\n", __func__);
 
        ether = netdev_priv(dev);
 
@@ -644,9 +645,11 @@ static int nuc970_ether_start_xmit(struct sk_buff *skb, struct net_device *dev)
 {
        struct nuc970_ether *ether = netdev_priv(dev);
        struct nuc970_txbd *txbd;
+       printk("%s\n", __func__);
 
        txbd = ether->tdesc + ether->cur_tx;
        if(txbd->mode & TX_OWEN_DMA) {
+               printk("%s 1\n", __func__);
                netif_stop_queue(dev);
                return NETDEV_TX_BUSY;
        }
@@ -667,6 +670,7 @@ static int nuc970_ether_start_xmit(struct sk_buff *skb, struct net_device *dev)
        txbd = ether->tdesc + ether->cur_tx;
        if(txbd->mode & TX_OWEN_DMA) {
                netif_stop_queue(dev);
+               printk("%s 2\n", __func__);
                //return NETDEV_TX_BUSY;
:
```

qianfan-Zhao, 2019-04-24:
Seems this can solve this problem, I will do more test about this issue.

```
diff --git a/drivers/net/ethernet/nuvoton/nuc970_ether0.c b/drivers/net/ethernet/nuvoton/nuc970_ether0.c
index d53f04bb7b7..cc8df9e9a04 100755
--- a/drivers/net/ethernet/nuvoton/nuc970_ether0.c
+++ b/drivers/net/ethernet/nuvoton/nuc970_ether0.c
@@ -456,6 +456,9 @@ static void nuc970_return_default_idle(struct net_device *dev)
        val = __raw_readl( REG_MCMDR);
        val |= SWR;
        __raw_writel(val,  REG_MCMDR);
+
+       while (__raw_readl(REG_MCMDR) & SWR)
+               cpu_relax();
 }
```

yachen, 2019-04-25:
Hi,

Clear SW should take effect immediately. This is working on HCLK domain.

Sincerely,

Yi-An Chen

The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

qianfan-Zhao, 2019-04-25:
Hi can you reproduce this problem?

qianfan-Zhao, 2019-04-25:
@yachen this patch can't solve the problem.  what's your opinion about this, software or hardware issue?

yachen, 2019-04-25:
Hi,

We’ve seen similar issue before, could be software or hardware issue.
By software: Enable LCD in U-Boot but disable LCD in Linux kernel.
By hardware: Some PHY is unstable, and can only bring back to normal by power off/on the PHY, a reset cannot bring the PHY back to normal.

Sincerely,

Yi-An Chen

qianfan-Zhao, 2019-04-25:
My board has a ti dp83822 phy and  the linux kernel can detect cable plugin and plugged when this issue happens. So i don't think this is a bug of PHY.

yachen, 2019-04-25:
Hi,

Please check the REFCLK (50MHz) signal and U-Boot LCD configuration. Both could cause Tx/Rx interrupt count stays at 0.

Sincerely,

Yi-An Chen

qianfan-Zhao, 2019-04-25:
thanks for yours guide and i will do a test

qianfan-Zhao, 2019-04-25:
@yachen Hi I had checked the hardware and the 50M refclock is OK, reset PHY doesn't make the network work again.

I will check the LCD configuration in uboot.

qianfan-Zhao, 2019-04-25:
@yachen do you have any idea about why enable LCD in uboot could cause network failed?

yachen, 2019-04-25:
Yes we know the reason.
LCD and EMAC0 are connect to the same AHB bus.
And LCD can only be disabled in certain time window otherwise it can lock itself and every peripheral on this AHB bus.

Sincerely,

Yi-An Chen

qianfan-Zhao, 2019-04-25:
@yachen What's the correct time window for disable the LCD? Is there has other documents for this?

yachen, 2019-04-25:
When LCD is not fetching data from DDR.
For example right after VSYNC(DCCS[28]) set 1.

Sincerely,

Yi-An Chen

qianfan-Zhao, 2019-04-25:
@yachen 
> When LCD is not fetching data from DDR. For example right after VSYNC(DCCS[28]) set 1.

If so, means if I enabled LCD in uboot but disabled in linux, this will reproduce this issue 100% ?

I tested this PR (waiting reset done) on multiple boards, but the result is strange. 2 boards seems got fixed and the other two still have this problem.

yachen, 2019-04-25:
Hi,

No. Not 100%, LCD is not always fetching data from DDR out of VSYNC period.
Is there a panel on your system?

Sincerely,

Yi-An Chen

qianfan-Zhao, 2019-04-25:
No, we don't have a LCD on board. 
I use the default u-boot configuration and with LCD enabaled(nuc972_defconfig enabled lcd by default).

I am testing u-boot with LCD disabled.

yachen, 2019-04-25:
So you're still seeing EMAC not functioning after disable LCD driver in U-Boot?
Can you dump register at 0xF0002200, 0xF0002204, 0xF0002208, and 0xF000220C when the problem occurs? Thanks. (Note: These are virtual address in kernel space)

Sincerely,

Yi-An Chen

qianfan-Zhao, 2019-04-25:
I am testing now, seems it is fixed on my board. 

I will tested this on the others boards and see if this issue is really fixed.

qianfan-Zhao, 2019-04-27:
the issue doesn't happen again when i remove CONFIG_LCD in uboot.

yachen, 2019-04-29:
No problem. I'll close this issue since it's solved.

closed 2019-04-29 by yachen