OpenNuvoton/NUC970_Linux_Kernel #41: Ethernet broken issue

jiuzhuaxiong, 2019-08-13:
when using rz command to transfer big file  to mounted nand flash or mounted U disk exfat file
system through ssh (scp), the ethernet will down , and restore in 10 seconds , not only ethernet , you will heard the usb cdc disconnect sound from computer.

yachen, 2019-08-14:
Hi,

About the USB CDC, do you mean the USB gadget of NUC970 or other VCOM converter?
Can you add printk in adjust_link() function of nuc970_ether0.c and see if there's phydev->link == 0 (link down event) or status_change == true (link speed change) during your test? Thanks. 

Sincerely,

Yi-An Chen

jiuzhuaxiong, 2019-08-16:
I will create the experiment again following your instructions , and give you feedback

closed 2019-10-21 by yachen