OpenNuvoton/NUC970_Linux_Kernel #45: watchdog doesn't reboot issue when timeout

jiuzhuaxiong, 2019-08-29:
when no feed watchdog , the system will hang, and does not reboot

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <unistd.h>

#include <fcntl.h>
#include <sys/ioctl.h>
#include <linux/types.h>
#include <linux/watchdog.h>
#include <signal.h>

int fd;

void catch_int(int signum)
{
    signal(SIGINT, catch_int);

    printf("In signal handler\n");
    if(0 != close(fd))
    {
        printf("Close failed in signal handler\n");
    }
    else
    {
        printf("Close succeeded in signal handler\n");
    }
}

int main(int argc,char* argv[])
{
    int ii;
    int timeout, timeout_bk;
    //signal(SIGINT, catch_int);
    fd = open("/dev/watchdog", O_RDWR);
    if (fd == -1)
    {
        perror("watchdog");
        exit(EXIT_FAILURE);
    }
    
    if(argc == 3)
    {
        timeout = atoi(argv[1]);
        ii = atoi(argv[2]);
    }
    else
    {
        timeout = 10;
        ii = 10;
    }
    ioctl(fd, WDIOC_GETTIMEOUT, &timeout_bk);
    printf("-Open watchdog ok;org timeout=%d\n", timeout_bk);

    ioctl(fd, WDIOC_SETTIMEOUT, &timeout);
    printf("The timeout is set to %d seconds\n", timeout);
        
    // Ping WDT for 10 times, then let system reset
    while (1)
    {
        printf("ii = %d\n", ii);
        if(ii-- > 0)
        {
            ioctl(fd, WDIOC_KEEPALIVE, 0);
            sleep(1);
        }
        else if (timeout-- > -5)
        {
			
            printf("timeout=%d\n", timeout);
            sleep(1);
        }
    }
    close(fd);
    return 0;
}

jiuzhuaxiong, 2019-08-29:
but you can call reboot , reboot the device .

yachen, 2019-08-30:
Hi,

Please make sure power on setting enables WDT (PA.3), otherwise WDT cannot reboot the system.

Sincerely,

Yi-An Chen

jiuzhuaxiong, 2019-08-31:
YES

jiuzhuaxiong, 2019-09-08:
Now I get the PA.3 to NC state , and get the watchdog=on, it works , but only enable it 2s timeout , how can i get this timeout value to 8.03 maximum value ?

yachen, 2019-09-09:
Set a timeout value equal to or larger than 8 sec with ioctl() command.

	if(timeout < 2) {
		val |= 0x5 << 8;
	} else if (timeout < 8) {
		val |= 0x6 << 8;
	} else {
		val |= 0x7 << 8;
	}

closed 2019-10-21 by yachen