OpenNuvoton/NUC970_Linux_Kernel #51: Kernel panic when connecting a bluetooth device

kormilicinkostia, 2020-07-06:
Good day.
I use nuc977DK62Y with the rtl8723bu chip.
I am trying to connect from my device via bt to another device.
In order to subscribe, I do the following:

```
hciconfig hci0 up
hcitool scan
rfcomm connect rfcomm0 00:19:86:00:29:7D &
```
When I try to connect, I see Linux kernel panic (I use the last commit from this repository):

```
`[  151.590000] rtk_btusb: hci0 evt 2
[  151.590000] rtk_btusb: btusb_notify : hci0 evt 2
[  151.590000] rtk_btusb: notify wifi start page
[  152.770000] rtk_btusb: notify wifi page success end
[  152.830000] rtk_btusb: L2CAP_CONNECTION_REQ, handle=2, PSM=3, scid=40
[  152.880000] rtk_btusb: L2CAP_CONNECTION_RESP, handle=2, dcid=0, scid=40, result=1
[  152.880000] rtk_btusb: L2CAP_CONNECTION_RESP, handle=2, dcid=40, scid=40, result=0
[  152.880000] rtk_btusb: l2cap connection success, update connection
[  152.880000] rtk_btusb: update_profile_connection: is_add=1, profile_index=3
[  152.880000] rtk_btusb: update_profile_connection: usb_coex_info.profile_bitmap = 8
[  152.880000] rtk_btusb: update_profile_connection: usb_coex_info.profile_refcount[0] = 0
[  152.880000] rtk_btusb: update_profile_connection: usb_coex_info.profile_refcount[1] = 0
[  152.880000] rtk_btusb: update_profile_connection: usb_coex_info.profile_refcount[2] = 0
[  152.880000] rtk_btusb: update_profile_connection: usb_coex_info.profile_refcount[3] = 1
[  152.880000] rtk_btusb: update_profile_connection: usb_coex_info.profile_refcount[4] = 0
[  152.880000] rtk_btusb: update_profile_connection: usb_coex_info.profile_refcount[5] = 0
[  152.880000] rtk_btusb: update_profile_connection: usb_coex_info.profile_refcount[6] = 0
[  152.880000] rtk_btusb: update_profile_connection: usb_coex_info.profile_refcount[7] = 0
[  152.880000] rtk_btusb: rtk_notify_profileinfo_to_fw, BufferSize is 5
[  152.880000] rtk_btusb: rtk_notify_profileinfo_to_fw, NumberOfHandles is 1
[  152.880000] rtk_btusb: rtk_notify_profileinfo_to_fw, handle is 2
[  152.880000] rtk_btusb: rtk_notify_profileinfo_to_fw, profile_bitmap is 8
[  152.880000] rtk_btusb: rtk_notify_profileinfo_to_fw, profile_status is 0
[  152.880000] rtk_btusb: rtk_vendor_cmd_to_fw, opcode is 0xfc19
[  153.020000] rtk_btusb: in hci_cmd_task, coex_info.num_hci_cmd_packet is 1
[  153.030000] rtk_btusb: send cmd to fw, opcode = 0xfc19
[  153.150000] Unable to handle kernel paging request at virtual address ffffffe0
[  153.160000] pgd = c0004000
[  153.160000] [ffffffe0] *pgd=03ffd831, *pte=00000000[  153.170000] Unable to handle kernel paging request at virtual address ffffffe0
[  153.170000] pgd = c3008000

[  153.190000] [ffffffe0] *pgd=03ffd831, *pte=00000000, *ppte=00000000
[  153.190000] Internal error: Oops: 17 [#1] PREEMPT ARM
[  153.190000] Modules linked in:
[  153.190000] CPU: 0 PID: 1071 Comm: klogd Not tainted 3.10.108+ #3
[  153.190000] task: c3865900 ti: c3be8000 task.ti: c3be8000
[  153.190000] PC is at __kmalloc_track_caller+0xa4/0x138
[  153.190000] LR is at __kmalloc_track_caller+0x6c/0x138
[  153.190000] pc : [<c00b0984>]    lr : [<c00b094c>]    psr: a0000013
[  153.190000] sp : c3be9cd0  ip : c3be9cd0  fp : c3be9cf4
[  153.190000] r10: 00000064  r9 : c3be8000  r8 : c030b458
[  153.190000] r7 : 000010da  r6 : 000106d0  r5 : c3801c80  r4 : ffffffe0
[  153.190000] r3 : 00000000  r2 : 00000001  r1 : 00000001  r0 : 00000000
[  153.190000] Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  153.190000] Control: 0005317f  Table: 03008000  DAC: 00000015
[  153.190000] Process klogd (pid: 1071, stack limit = 0xc3be81b8)
[  153.190000] Stack: (0xc3be9cd0 to 0xc3bea000)
[  153.190000] 9cc0:                                     c3ab0d80 00000100 000004d0 c3801d00
[  153.190000] 9ce0: 00000000 c030b458 c3be9d24 c3be9cf8 c0310ed4 c00b08f0 ffffffff c3bce1c0
[  153.190000] 9d00: 000004d0 00000000 00000000 00000000 c3be9d40 00000064 c3be9d74 c3be9d28
[  153.190000] 9d20: c030b458 c0310e6c c00426e0 00000039 000000c3 c03136f0 c3be9d5c c3bce000
[  153.190000] 9d40: 00000001 c3ab0d80 00000100 c3bce000 c3bce1c0 00000000 00000039 c3be9ecc
[  153.190000] 9d60: 00000000 c3be9df8 c3be9df4 c3be9d78 c0379470 c030b3bc c3be9dac c34aadd8
[  153.190000] 9d80: c3be9dd4 c0579db8 c3be9da4 c3be9d98 00000000 c3403dc0 00000039 00000000
[  153.190000] 9da0: 0000001d c3bf7aa8 c3be9dd4 ffffffa6 c00cec04 00000000 00000000 00000000
[  153.190000] 9dc0: ffffffff ffffffff c05c3d60 c3403dc0 00000039 c3be9ecc c3865900 bed21898
[  153.190000] 9de0: 00000000 00000000 c3be9eb4 c3be9df8 c0307b24 c0379300 c3bf7aa0 000005b4
[... 46 lines cut ...]
[  153.600000] Stack: (0xc306ff10 to 0xc3070000)
[  153.600000] ff00:                                     c307f480 00000000 c306e038 c05690cc
[  153.600000] ff20: 00000000 c306e000 c306ff7c c306ff38 c00ba424 c00aead0 c003f97c c006ed7c
[  153.600000] ff40: c3b47000 c305fc64 00000000 c003f930 ffffffff c307f480 c30c8e40 c306e000
[  153.600000] ff60: 00000000 00000000 00000000 00000000 c306ff9c c306ff80 c0030da0 c00ba3a8
[  153.600000] ff80: c05a2414 c30c8e40 c0030dc4 00000000 c306ffac c306ffa0 c0030dec c0030c84
[  153.600000] ffa0: 00000000 c306ffb0 c000e2d8 c0030dd4 00000000 00000000 00000000 00000000
[  153.600000] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  153.600000] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[  153.600000] Backtrace:
[  153.600000] [<c00aeac0>] (kmem_cache_alloc+0x0/0x104) from [<c00ba424>] (do_execve+0x8c/0x490)
[  153.600000]  r9:c306e000 r8:00000000 r7:c05690cc r6:c306e038 r5:00000000
r4:c307f480
[  153.600000] [<c00ba398>] (do_execve+0x0/0x490) from [<c0030da0>] (____call_usermodehelper+0x12c/0x150)
[  153.600000] [<c0030c74>] (____call_usermodehelper+0x0/0x150) from [<c0030dec>] (call_helper+0x28/0x30)
[  153.600000]  r6:00000000 r5:c0030dc4 r4:c30c8e40 r3:c05a2414
[  153.600000] [<c0030dc4>] (call_helper+0x0/0x30) from [<c000e2d8>] (ret_from_fork+0x14/0x3c)
[  153.600000] Code: ebfffe58 e1a04000 ea000017 e5953014 (e7942003)
[  153.820000] ---[ end trace bc1d8fbff08e3bae ]---
Connected /dev/rfcomm0 to 00:19:86:00:29:7D on channel 1`
```

Using gdb, I found out that an error occurs in the /mm/slub.c file in the get_freepointer function, presumably due to an invalid pointer to the kmem_cache structure
In order to connect the bt device, I assembled the driver provided by the vendor (if necessary, I can transfer it to you). I checked that it wasn’t the driver because I built this driver for my raspberry pi and it works successfully there.
Can you help me solve this problem?

kormilicinkostia, 2020-07-30:
@yachen

yachen, 2020-07-31:
Hi, I'm afraid I don't have the environment to check this issue. We're not seeing such panic using RTL WiFi dongles which we have.

closed 2021-11-23 by ychuang3