OpenNuvoton/NUC970_Linux_Kernel #55: I have defined pi10-pi14 as the LED GPIO port.

junwushi, 2020-09-09:
The following error occurred：
leds-gpio leds-gpio: pins are not configured from the driver，

Should I specify them in pincrtl?
 I don't know how to define those structures in pinctl, can help me some sample for it?

yachen, 2020-09-09:
Hi,

Are you using device tree? If so, pleas try:
```
        gpio_leds_test {
                compatible = "gpio-leds";
                led0 {
                        label = "GPA2 LED";
                        gpios = <&gpio 2  1 /*GPIO_ACTIVE_LOW*/>;
                };
                led1 {
                        label = "GPD2 LED";
                        gpios = <&gpio 98 0 /*GPIO_ACTIVE_HIGH*/>;
                };
        };
```
Or  you can use sysfs to control GPIO. For example:
PA0:
```
$ echo 0 > /sys/class/gpio/export
$ echo out >/sys/class/gpio/gpio0/direction
$ echo 1 >/sys/class/gpio/gpio0/value
```
PD2:
```
$ echo 98 > /sys/class/gpio/export
$ echo out >/sys/class/gpio/gpio98/direction
$ echo 1 >/sys/class/gpio/gpio98/value
```

Sincerely,

Yi-An Chen

junwushi, 2020-09-10:
i am not use device tree. it is define in dev.c
#if defined(CONFIG_LEDS_GPIO)
static struct gpio_led gpio_leds[] = {  
    {  
        .name = "power",  
        .gpio = NUC970_PI12,
        .default_state = LEDS_GPIO_DEFSTATE_ON, // 默认LED亮  
        .active_low = 0, // 低电平亮  
      //  .default_trigger = "timer", // 触发器  
    },  
    {  
        .name = "tx",  
        .gpio = NUC970_PI14,
        .default_state = LEDS_GPIO_DEFSTATE_OFF, 
        .active_low = 0, // 低电平亮  
     //   .default_trigger = "timer", // 触发器  
    },  
    {  
        .name = "rx",  
        .gpio = NUC970_PI11,
        .default_state = LEDS_GPIO_DEFSTATE_OFF, 
        .active_low = 0, // 低电平亮  
    //    .default_trigger = "timer", // 触发器  
    },  
    {  
        .name = "err",  
        .gpio = NUC970_PI13,
        .default_state = LEDS_GPIO_DEFSTATE_OFF, 
        .active_low = 0, // 低电平亮  
  //      .default_trigger = "timer", // 触发器  
    },  
    {  
        .name = "online",  
        .gpio = NUC970_PI10,
        .default_state = LEDS_GPIO_DEFSTATE_OFF, 
        .active_low = 0, // 低电平亮  
//        .default_trigger = "timer", // 触发器  
    },  
};

schung1218, 2020-09-10:
For example:  dev.c

#include <linux/leds.h>
static struct gpio_led def_leds[] = {
{
.name = "power",
.gpio = NUC970_PI12,
.default_state = LEDS_GPIO_DEFSTATE_ON, // 默认LED亮
.active_low = 0, // 低电平亮
// .default_trigger = "timer", // 触发器
},
{
.name = "tx",
.gpio = NUC970_PI14,
.default_state = LEDS_GPIO_DEFSTATE_OFF,
.active_low = 0, // 低电平亮
// .default_trigger = "timer", // 触发器
},
{
.name = "rx",
.gpio = NUC970_PI11,
.default_state = LEDS_GPIO_DEFSTATE_OFF,
.active_low = 0, // 低电平亮
// .default_trigger = "timer", // 触发器
},
{
.name = "err",
.gpio = NUC970_PI13,
.default_state = LEDS_GPIO_DEFSTATE_OFF,
.active_low = 0, // 低电平亮
// .default_trigger = "timer", // 触发器
},
{
.name = "online",
.gpio = NUC970_PI10,
.default_state = LEDS_GPIO_DEFSTATE_OFF,
.active_low = 0, // 低电平亮
// .default_trigger = "timer", // 触发器
},
};

 static struct gpio_led_platform_data gpio_leds = {
  .num_leds = ARRAY_SIZE(def_leds),
  .leds = def_leds,
 };

static struct platform_device nuc970_gpio_leds_device = {
       .name                    = "leds-gpio",
       .id                 = -1,
       .dev.platform_data  = &gpio_leds,
};

static struct platform_device *nuc970_public_dev[] __initdata = {
        &nuc970_serial_device0,
        &nuc970_gpio_leds_device,

...

junwushi, 2020-09-10:
thanks schung128, I already do it, 
but the following error occurred when kernel start：
leds-gpio leds-gpio: pins are not configured from the driver，
wether gpio not initialization，It should also be defined in pinctrl driver？

schung1218, 2020-09-10:
@junwushi 
Because all PINs default values are GPIO, so you don’t need to use pinctrl API, so please to try
" Remove devm_pinctrl_get_select_default "

closed 2020-09-11 by junwushi