OpenNuvoton/NUC970_Linux_Kernel #56: How to turn on hardware random accelerator

chenxy1988, 2020-09-10:
Hi Nuvoton team,

We met a problem that the entropy increased too slow to cause openssl (> 1.1.1d) read /dev/random successfully. I used "cat /proc/sys/kernel/random/entropy_avail" to observed the entropy is less than 100 when power on the board. 
I noticed that kernel has an option "CONFIG_HW_RANDOM_NUVOTON", but seems related TMP. Does this driver can do random accelerator without TMP?

Br,
Xiangyu

yachen, 2020-09-10:
Hi,

The option is for enable the built in hardware random number generator. It has nothing to do with TPM.
Your system can get random number by reading /dev/hwrng.

Sincerely,

Yi-An Chen

chenxy1988, 2020-09-10:
Thanks. Closing this ticket.

closed 2020-09-10 by chenxy1988