OpenNuvoton/NUC970_Linux_Kernel #57: Question on NUC970 LCD controller clock

chenxy1988, 2020-10-27:
Hi Nuvoton team,

We used a LCD (AT070TN92 800*480) on NXP platform before,  after migrating to NUC970 platform, we found some difference on LCD clock.

We set the lcd clock to 33Mhz to line up with datasheet typical value, the NXP platform is working, on NUC970 platform, we set 33300000 but the clock we measured is 30Mhz(lcd parameter code in the end of this thread).  Is there something need also to config? 

Could you share the formula of LCD clock calculate with us? Thanks!

The LCD parameter on arch/arm/mach-nuc970/dev.c   nuc970fb_lcd_info structure as below:
  [0] = {
421                 .type   = LCM_DCCS_VA_SRC_RGB565,
422                 .bpp            = 16,
423                 .width          = 800,
424                 .height         = 480,
425                 .xres           = 800,          /* Pixels per line */
426                 .yres           = 480,          /* Lines per panel */
427                 .pixclock       = 33300000,
428                 .left_margin    = 40,           /* Horizontal back porch */
429                 .right_margin   = 210,          /* Horizontal front porch */
430                 .hsync_len      = 1,            /* HSYNC pulse width */
431                 .upper_margin   = 23,           /* Vertical back porch */
432                 .lower_margin   = 22,           /* Vertical front porch */
433                 .vsync_len      = 1,            /* VSYNC pulse width */
434                 .dccs           = 0x0e00040a,
435                 .fbctrl         = 0x01900190,
436                 .devctl         = 0x050000c0,
437                 .scale          = 0x04000400,
438         },

Thanks!

Xiangyu

chenxy1988, 2020-10-27:
Attached datasheet of LCD manual url :http://www.tekswo-display.com/tools/download.ashx?id=25

yachen, 2020-10-27:
Hi @chenxy1988 , 

The setting formula is CLKSRC/(DIV +１), where CLKSRC is UPLL (300MHz) in Linux kernel. 
You're seeing output clock set to 30000000 instead of 33333333 (CLKSRC / 9) is caused by the round function in driver/clk/clk-divider.c. Unlike later kernel that provides round to closest option, it can only round up to next divider.

If you need to set the pixel clock near 33MHz, please set .pixclock  to 33333334. The result will be 33333333 Hz.

Sincerely,

Yi-An Chen

chenxy1988, 2020-10-28:
Thanks Yi An,
I have tried and the clock set to 33.3Mhz successfully!

Confirm and close this issue.

Thanks!

closed 2020-10-28 by chenxy1988