OpenNuvoton/NUC970_Linux_Kernel #59: Is this possible add a msleep in loop to avoid watchdog was fired?

chenxy1988, 2020-11-30:
Hi Nuvoton team,

The section of the code has a loop that polling a register to detect the value was written correct or not. We found that this would entry a dead loop and trigger the watchdog timeout in some corner case.
Is this possible to add a msleep like function to give up some CPU time for other task schdule when polling the register?

https://github.com/OpenNuvoton/NUC970_Linux_Kernel/blob/f6fd11eee272e6eb12892974356c11066e12b2d8/drivers/usb/gadget/nuc970_udc.c#L1894

Thanks
Xiangyu

yachen, 2020-11-30:
Hi,

This loop is used to wait until USB PHY is feeding clock into USB device. USB gadget shouldn't perform reset of the 
initialize code unless there's stable clock feed from PHY. And we don't expect the kernel to be trapped here.

Anyway, can you see if adding a cond_resched() call in this loop fix this issue? Thanks.

Sincerely,

Yi-An Chen

chenxy1988, 2020-11-30:
Thanks Yi An,  I'll try that. Now we are treating the usb host and gadget as kernel module and install and unstall it when we want to the target board play usb host/device role. It would cause wdt reset sometimes when we install gadget module and  not 100% reproduce.

closed 2020-11-30 by chenxy1988

chenxy1988, 2020-12-01:
I have tried add cond_resched() in the loop, and didnt see the wdt was triggered anymore during insmod/rmmod gadget driver kernel module

yachen, 2020-12-01:
Thanks. We'll add the cond_resched() into this loop.