OpenNuvoton/NUC970_Linux_Kernel #60: Does nuc970 watchdog has some trace information?

chenxy1988, 2021-01-05:
Hi Nuc team,

Recently, we found our product random reboot without any action, in order to spilt the root cause in application layer kernel layer or hardware layer(hope not :p), we have to add or enable debug trace when wdt is timeout, is there any way to do that?

Thanks!

Br,
Xiangyu

yachen, 2021-01-05:
Not sure what do you mean by the "debug trace". You can dump the register SYS_RSTSTS for the cause of system reset source.

chenxy1988, 2021-01-12:
Regarding SYS_RSTSTS, 
normally, system boot up, the value is 0x7, input "reboot" in linux, after boot complate, this value also 0x7. 
Kill the watchdog, system reset by wdt, after boot complate, this value is 0x27, input the "reboot" command, after boot complate, this value still 0x27 without change.
My question is does "reboot" command in linux no impact on SYS_RSTSTS register value?

Thanks.

yachen, 2021-01-12:
Hi,

In order to get the latest reset source, please write 1s to clear this register after read.

Sincerely,

Yi-An Chen

chenxy1988, 2021-01-13:
Many thanks! It's working correctly now ;)

closed 2021-01-13 by chenxy1988