OpenNuvoton/NUC970_Linux_Kernel #61: Error while using make command

tkrelcon, 2021-03-08:
Dear Sir,

Can help out in below error as i am confused a little in this.

"**/usr/local/arm_linux_4.8/usr/bin/make: cannot execute binary file: Exec format error** "

yachen, 2021-03-08:
That's an ARM executable. Are you executing it on a x86 machine?

tkrelcon, 2021-03-08:
Please find below screen shots of Host window machine and Virtual Ubuntu. I am using Linux VDI downloaded from Nuvoton website only.
[image]
[image]

yachen, 2021-03-08:
Please type "file /usr/local/arm_linux_4.8/usr/bin/make" to check the file format. It's a ARM executable (at least the file on my notebook is), so it cannot execute on X86 system.

tkrelcon, 2021-03-08:
Yes sir, It is. 

user@ubuntu:~$ file /usr/local/arm_linux_4.8/usr/bin/make
/usr/local/arm_linux_4.8/usr/bin/make: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-, with debug_info, not stripped
user@ubuntu:~$

tkrelcon, 2021-03-08:
How to remove this error?

yachen, 2021-03-08:
You can't. This is a executable built for execute on an ARM CPU, but you're trying to execute it on a X86 machine.

tkrelcon, 2021-03-08:
So for this i need 64 bit Linux OS ?

yachen, 2021-03-08:
No. The "make" you're trying to execute is compile to ARM machine code.
X86 machine code is not execute ARM machine code. If you need a make for X86 Ubuntu, please use apt to download the correct version.

tkrelcon, 2021-03-08:
Can you suggest further steps? Please. I am new for Ubuntu and Linux..

closed 2021-03-24 by tkrelcon