OpenNuvoton/NUC970_Linux_Kernel #62: NUC972 Board Ethernet not working

tkrelcon, 2021-03-09:


tkrelcon, 2021-03-09:
# ifconfig -a
eth0      Link encap:Ethernet  HWaddr 08:00:27:00:01:92
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

yachen, 2021-03-09:
What board are you using? Is there any error message during system boot up?

Sincerely,

Yi-An Chen

________________________________
The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

tkrelcon, 2021-03-09:
Enable Debug Message
No Key
Boot from NAND
NAND boot!

U-Boot 2016.11 (Mar 09 2021 - 14:57:39 +0530)

CPU: NUC972
Board: NUC972
DRAM:  64 MiB
NAND:  128 MiB
In:    serial
Out:   serial
Err:   serial
Net:   emac
**Error: emac address not set.**

Hit any key to stop autoboot:  0
Unknown command 'sf' - try 'help'
Unknown command 'sf' - try 'help'
Wrong Image Format for bootm command
ERROR: can't get kernel image!
=> nboot 0x7fc0 0 0x200000

Loading from nand0, offset 0x200000
   Image Name:
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3939744 Bytes = 3.8 MiB
   Load Address: 00007fc0
   Entry Point:  00008000
=> bootm 0x7fc0
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3939744 Bytes = 3.8 MiB
   Load Address: 00007fc0
   Entry Point:  00008000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK

Starti▒Booting Linux on physical CPU 0x0
Initializing cgroup subsys cpuset
Initializing cgroup subsys cpu
Initializing cgroup subsys cpuacct
Linux version 3.10.108 (user@ubuntu) (gcc version 4.8.5 (Buildroot 2016.11.1-g7116cea) ) #2 PREEMPT Tue Mar 9 15:08:23 IST 2021
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC970
Memory policy: ECC disabled, Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: root=/dev/ram0 console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 64MB = 64MB total
Memory: 59000k/59000k available, 6536k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    vmalloc : 0xc4800000 - 0xff000000   ( 936 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc0339eec   (3272 kB)
      .init : 0xc033a000 - 0xc056174c   (2206 kB)
      .data : 0xc0562000 - 0xc0595560   ( 206 kB)
       .bss : 0xc0595560 - 0xc05c54e0   ( 192 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Preemptible hierarchical RCU implementation.
NR_IRQS:625
=============================================================================
[... 59 lines cut ...]
Starting logging: OK
Starting mdev...
Initializing random number generator... done.
Starting network: OK
# ifconfig -a
eth0      Link encap:Ethernet  HWaddr 08:00:27:00:01:92
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

# ip link show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop qlen 1000
    link/ether 08:00:27:00:01:92 brd ff:ff:ff:ff:ff:ff

tkrelcon, 2021-03-10:
Any help or solution??

yachen, 2021-03-10:
You didn't turn off LCD in U-Boot.

WARNING: LCD enabled in loader but not in kernel, this may cause EMAC hang...

Sincerely,

Yi-An Chen
________________________________
The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

tkrelcon, 2021-03-12:
Thank you sir, I got it but how turn it off. I done it in nuc970evb.h file but the same error.  Also i was making nuc972_eth2uart_defconfig where it is already deisabled.
Please guide which file to edit by which parameter.

tkrelcon, 2021-03-12:
I think now i have successfully turned it off.. But still facing same problem.. Please guide. Refer below lines of output.

Loading from nand0, offset 0x200000
   Image Name:
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4169728 Bytes = 4 MiB
   Load Address: 00007fc0
   Entry Point:  00008000
=> bootm 0x7fc0
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4169728 Bytes = 4 MiB
   Load Address: 00007fc0
   Entry Point:  00008000
   Verifying Checksum ... OK
   Loading Kernel Image ... OK

Startic▒ѥ▒▒ Linux on physical CPU 0x0
Initializing cgroup subsys cpuset
Initializing cgroup subsys cpu
Initializing cgroup subsys cpuacct
Linux version 3.10.108 (user@ubuntu) (gcc version 4.8.5 (Buildroot 2016.11.1-g7116cea) ) #8 PREEMPT Fri Mar 12 09:53:22 IST 2021
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC970
Memory policy: ECC disabled, Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: root=/dev/ram0 console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 64MB = 64MB total
Memory: 58760k/58760k available, 6776k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    vmalloc : 0xc4800000 - 0xff000000   ( 936 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc033feec   (3296 kB)
      .init : 0xc0340000 - 0xc059c9f4   (2419 kB)
      .data : 0xc059e000 - 0xc05d1e40   ( 208 kB)
       .bss : 0xc05d1e40 - 0xc0601dc0   ( 192 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Preemptible hierarchical RCU implementation.
NR_IRQS:625
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 4294967286ms
Console: colour dummy device 80x30
console [ttyS0] enabled
Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 512
Initializing cgroup subsys devices
Initializing cgroup subsys freezer
Initializing cgroup subsys perf_event
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0xc0296838 - 0xc0296874
devtmpfs: initialized
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
bio: create slab <bio-0> at 0
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
cfg80211: Calling CRDA to update world regulatory domain
NET: Registered protocol family 2
TCP established hash table entries: 512 (order: 0, 4096 bytes)
[... 45 lines cut ...]
nuc970-emac0 nuc970-emac0: eth0 is OPENED
ifconfig: SIOCSIFADDR: No such device

~ # ifconfig -a
eth0      Link encap:Ethernet  HWaddr 08:00:27:00:01:92
          inet addr:192.168.0.100  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          LOOPBACK  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tunl0     Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          NOARP  MTU:0  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

yachen, 2021-03-12:
Hi,

The kernel log looks good. Which board are you using?

Sincerely,

Yi-An Chen
________________________________
The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

tkrelcon, 2021-03-12:
Dear Sir, Please find below pics

[image]
[image]

yachen, 2021-03-12:
Have you set SW21 and SW38 to ON?

Sincerely,

Yi-An Chen
________________________________
The privileged confidential information contained in this email is intended for use only by the addressees as indicated by the original sender of this email. If you are not the addressee indicated in this email or are not responsible for delivery of the email to such a person, please kindly reply to the sender indicating this fact and delete all copies of it from your computer and network server immediately. Your cooperation is highly appreciated. It is advised that any unauthorized use of confidential information of Nuvoton is strictly prohibited; and any information in this email irrelevant to the official business of Nuvoton shall be deemed as neither given nor endorsed by Nuvoton.

tkrelcon, 2021-03-12:
Yes both are on already

tkrelcon, 2021-03-12:
Any suggestion/help further?

yachen, 2021-03-12:
Please test attached kernel image. If the SW21/SW38 is ON and EMAC0 still doesn’t work, I would guess there’re something wrong about your board.

[970uimage.zip](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/6131102/970uimage.zip)

tkrelcon, 2021-03-13:
U-Boot 2013.04-rc2-ga8435de-dirty (Apr 18 2016 - 10:47:35)

CPU: NUC972
DRAM:  64 MiB
NAND:  128 MiB
MMC:   mmc: 0
In:    serial
Out:   serial
Err:   serial
Net:   emac
U-Boot> nboot 0x7fc0 0 0x200000

Loading from nand0, offset 0x200000
   Image Name:   Linux-3.10.108+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4254928 Bytes = 4.1 MiB
   Load Address: 00008000
   Entry Point:  00008000
U-Boot> bootm 0x7fc0
## Booting kernel from Legacy Image at 00007fc0 ...
   Image Name:   Linux-3.10.108+
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4254928 Bytes = 4.1 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
   XIP Kernel Image ... OK
OK

Starting kernel ...

Uncompressing Li done, booting tBooting Linux on physical CPU 0x0
Initializing cgroup subsys cpuset
Initializing cgroup subsys cpu
Initializing cgroup subsys cpuacct
Linux version 3.10.108+ (andy@YA-T470P) (gcc version 4.8.4 (GCC) ) #144 PREEMPT Fri Mar 12 17:16:08 CST 2021
CPU: ARM926EJ-S [41069265] revision 5 (ARMv5TEJ), cr=00053177
CPU: VIVT data cache, VIVT instruction cache
Machine: NUC970
Memory policy: ECC disabled, Data cache writeback
Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
Kernel command line: root=/dev/ram0 console=ttyS0,115200n8 rdinit=/sbin/init mem=64M
PID hash table entries: 256 (order: -2, 1024 bytes)
Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
Memory: 64MB = 64MB total
Memory: 58680k/58680k available, 6856k reserved, 0K highmem
Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
    vmalloc : 0xc4800000 - 0xff000000   ( 936 MB)
    lowmem  : 0xc0000000 - 0xc4000000   (  64 MB)
    modules : 0xbf000000 - 0xc0000000   (  16 MB)
      .text : 0xc0008000 - 0xc033aeec   (3276 kB)
      .init : 0xc033b000 - 0xc05b005c   (2517 kB)
      .data : 0xc05b2000 - 0xc05e55c0   ( 206 kB)
       .bss : 0xc05e55c0 - 0xc06155c0   ( 192 kB)
SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Preemptible hierarchical RCU implementation.
NR_IRQS:625
sched_clock: 32 bits at 100 Hz, resolution 10000000ns, wraps every 4294967286ms
Console: colour dummy device 80x30
console [ttyS0] enabled
Calibrating delay loop... 148.88 BogoMIPS (lpj=744448)
pid_max: default: 32768 minimum: 301
Mount-cache hash table entries: 512
Initializing cgroup subsys devices
Initializing cgroup subsys freezer
Initializing cgroup subsys perf_event
CPU: Testing write buffer coherency: ok
Setting up static identity map for 0xc02917d8 - 0xc0291814
devtmpfs: initialized
pinctrl core: initialized pinctrl subsystem
NET: Registered protocol family 16
DMA: preallocated 256 KiB pool for atomic coherent allocations
bio: create slab <bio-0> at 0
SCSI subsystem initialized
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
cfg80211: Calling CRDA to update world regulatory domain
NET: Registered protocol family 2
TCP established hash table entries: 512 (order: 0, 4096 bytes)
TCP bind hash table entries: 512 (order: -1, 2048 bytes)
TCP: Hash tables configured (established 512 bind 512)
TCP: reno registered
UDP hash table entries: 256 (order: 0, 4096 bytes)
UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
NET: Registered protocol family 1
NetWinder Floating Point Emulator V0.97 (double precision)
ROMFS MTD (C) 2007 Red Hat, Inc.
msgmni has been set to 114
Block layer SCSI generic (bsg) driver version 0.4 loaded (major 254)
io scheduler noop registered
io scheduler deadline registered
io scheduler cfq registered (default)
nuc970-uart.0: ttyS0 at I/O 0x0 (irq = 36) is a NUC970
brd: module loaded
libphy: nuc970_rmii0: probed
ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
ehci_nuc970_probe() - name: nuc970-ehci
nuc970-ehci nuc970-ehci: Nuvoton NUC970/N9H30 EHCI Host Controller
nuc970-ehci nuc970-ehci: new USB bus registered, assigned bus number 1
nuc970-ehci nuc970-ehci: irq 23, io mem 0xb0005000
nuc970-ehci nuc970-ehci: USB 2.0 started, EHCI 0.95
hub 1-0:1.0: USB hub found
hub 1-0:1.0: 2 ports detected
ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
nuc970-ohci nuc970-ohci: Nuvoton NUC970/N9H30 OHCI Host Controller
nuc970-ohci nuc970-ohci: new USB bus registered, assigned bus number 2
nuc970-ohci nuc970-ohci: irq 24, io mem 0xb0007000
hub 2-0:1.0: USB hub found
hub 2-0:1.0: 2 ports detected
usbcore: registered new interface driver usb-storage
TCP: cubic registered
NET: Registered protocol family 17
Freeing unused kernel memory: 2516K

BusyBox v1.22.1 (2016-02-03 14:11:04 CST) built-in shell (ash)
Enter 'help' for a list of built-in commands.

~ # ifconfig -a
eth0      Link encap:Ethernet  HWaddr 08:00:27:00:01:92
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          LOOPBACK  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tkrelcon, 2021-03-13:
Sir, Please check above logs with image provided by you its giving me same output..
Any check points suggestion for hardware..

yachen, 2021-03-13:
Hi, did you ifconfig eth0 up?

Yi-An Chen

tkrelcon, 2021-03-13:
Yes Sir,

Please see below logs

~ # ifconfig eth0 up
nuc970-emac0 nuc970-emac0: eth0 is OPENED
~ # ifconfig -a
eth0      Link encap:Ethernet  HWaddr 08:00:27:00:01:92
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          LOOPBACK  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

yachen, 2021-03-13:
Are the LEDs on the Ethernet jack Light up?

Yi-An Chen

tkrelcon, 2021-03-13:
Yes Sir, Orange Continuously and green blinking. I check cable and pin both are working in other device properly.. 
while i am connecting with putty it is sending data and getting ping there.. I am not sure but something i am missing from software side.. I update everything from git also..

tkrelcon, 2021-03-13:
While connecting through putty using IP i am getting this....

login as: root
root@192.168.0.211's password:

Welcome to Automation BOS System..
root@Automation-BOS:~# ping 192.168.0.1
PING 192.168.0.1 (192.168.0.1) 56(84) bytes of data.
64 bytes from 192.168.0.1: icmp_seq=1 ttl=64 time=0.734 ms
64 bytes from 192.168.0.1: icmp_seq=2 ttl=64 time=0.646 ms
64 bytes from 192.168.0.1: icmp_seq=3 ttl=64 time=0.669 ms
64 bytes from 192.168.0.1: icmp_seq=4 ttl=64 time=0.670 ms
64 bytes from 192.168.0.1: icmp_seq=5 ttl=64 time=0.806 ms
64 bytes from 192.168.0.1: icmp_seq=6 ttl=64 time=0.678 ms
64 bytes from 192.168.0.1: icmp_seq=7 ttl=64 time=0.675 ms

tkrelcon, 2021-03-13:
This is what i am getting while using some networking commands

~ # traceroute 192.168.0.100
traceroute to 192.168.0.100 (192.168.0.100), 30 hops max, 38 byte packets
 1traceroute: sendto: Network is unreachable
~ # netstat
Active Internet connections (w/o servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
Active UNIX domain sockets (w/o servers)
Proto RefCnt Flags       Type       State         I-Node Path
~ # ipaddr
-/bin/sh: ipaddr: not found
~ # ip -s link
BusyBox v1.22.1 (2016-02-03 14:11:04 CST) multi-call binary.

Usage: ip [OPTIONS] {address | route | link | tunnel | rule} {COMMAND}

ip [OPTIONS] OBJECT {COMMAND}
where OBJECT := {address | route | link | tunnel | rule}
OPTIONS := { -f[amily] { inet | inet6 | link } | -o[neline] }

~ # arp -n
arp: applet not found

yachen, 2021-03-13:
What city are you in? I can't tell what goes wrong from your description. Do you have other 972 board in hand that you can give it a try?

Yi-An Chen

tkrelcon, 2021-03-15:
I am Vadodara(Baroda), Gujarat, India. No i don't have any other NUC970 board. 
We have used this board 2-3 years back. At that time everything was fine but at present do now whats going wrong. According to me there is some thing wrong with its image files somewhere it was blocking EMAC. Dn't know where. I will work on it let see if i get anything.. 
By the way thank you so much for your guidance. Have a good day ahead.

tkrelcon, 2021-03-15:
One more thing i have noticed while i am building through buildroot toolchain in environment file it was not setting any variable for ethernet. My environment Variables are as follows. Will it affect.?

baudrate=115200
bootdelay=1
stderr=serial
stdin=serial
stdout=serial
setspi=sf probe 0 30000000
loadkernel=sf read 0x7fc0 0x200000 0x800000
bootcmd=run setspi;run loadkernel;bootm 0x7fc0

yachen, 2021-03-15:
No, the Uboot env setting is irrelevant with the EMAC issue in Linux kernel.

tkrelcon, 2021-03-15:
Ok. Than I have to find something in kernel side only... Is there any other kernel configuration required for EMAC  Any changes required in Config files.?
 One more thing as per manual evb.h file should be this..
[image]

But in my files it is like this. If try do changes and than make, its giving me error in make.
#define CONFIG_ETHADDR: 00:00:00:11:66:88
#define CONFIG_SYS_RX_ETH_BUFFER: 16

yachen, 2021-03-15:
The original source doesn't have the extra colon. And can be built without problem.
https://github.com/OpenNuvoton/NUC970_U-Boot_v2016.11/blob/master/include/configs/nuc970_evb.h

tkrelcon, 2021-03-15:
Ok. I removed extra colon but nothing changed in previous one. Ok. Thank you again. If you get any solution idea pls let me know. I will search out the fault.

tkrelcon, 2021-03-24:
Hey, I got a new NUC980DK61YC Evaluation board with me. I am trying to load image files and uboot files in SPI but it was showing this error.

NUC980 IBR 20180813
Boot from USB
DDR-OK
=======================================
Run firmware code
CPU: 300MHz, DDR: 300MHz, SYS: 300MHz, PCLK: 150MHz
Parse NuWriter command line
=======================================
SPI NOR ID=0x0000ffff  _spi_type =0
SPI NOR ID=0x0000ffff  _spi_type =0
**Error! Read SPI ID(0xffff)**

SPI NAND: Auto Detect
ID=[0xefaa21]
BlockPerFlash = 1024, PagePerBlock = 64
PageSize = 2048, SpareArea = 64
QuadReadCmd    = 0x6b
ReadStatusCmd  = 0xff
WriteStatusCmd = 0xff
StatusValue    = 0xff
Dummybyte      = 0x1
Multi-Die      = 0

NAND ID=[57][57][57][57]
PowerOn setting 0x7f6002d4
BlockPerFlash=1023, PagePerBlock=64, PageSize=2048

eMMC1/SD1 Init Fail

Finish get INFO!!

closed 2021-03-25 by tkrelcon