OpenNuvoton/NUC970_Linux_Kernel #63: Console disconnected caused the application freeze

chenxy1988, 2021-04-07:
Dear NUC team,

Recently we met a issue related to debug console. In our product, the Uart0 as console port, and the console debug tool we use putty on PC side. The product has a LCD could be shown Qt GUI.

When we start an application with lots of debug output. Then system running well, we also can observe the data updating on lcd screen. When we disconnect the console on putty. we can observe the application was freeze on lcd until we reconnect the console on putty. 

Have you ever seen this behavior , is there any special configuration in 970 serial port driver?

Thanks.

Br,
Xiangyu

yachen, 2021-04-07:
Hi,

Is there a transceiver for the  console port?  Or the UART pins are floating while console is disconnected?

Sincerely,

Yi-An Chen

chenxy1988, 2021-04-07:
Hi Yi An,

There is a USB-232 converter, USB side connected PC and serial Tx Rx GND connected to target board. The converter chipset is FT232. 
According your reply, i did another test: just plug out the USB cable from PC instead of close putty. The application doesn't go to freeze.
I pluged in USB to pc again, close putty instead of disconnecting USB cable. The issue happens, application went freeze.

Both above operations, the UART pins are not floating.

Best regards,
Xiangyu

yachen, 2021-04-07:
Just a wild guess,  maybe the system seems blocked because there's huge amount of garbage coming from UART RX pin. Can you disconnect NUC970's UART0 RX pin with FT232 and test again?

chenxy1988, 2021-04-07:
Sure, I'll try to remove tx/rx pin when this issue happens.

chenxy1988, 2021-04-27:
Close due to after disconnected the FT232 with target board, the issue no longer happens anymore.
Thanks.

closed 2021-04-27 by chenxy1988