OpenNuvoton/NUC970_Linux_Kernel #66: nuc970_sd: cannot mount SD card on tomato

VS0x01, 2021-04-21:
[dmesg.log](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/6351461/dmesg.log)
Used tomato defconfig, added CONFIG_NUC970_SD0_EN=y. 
Without this kernel boots.

yachen, 2021-04-22:
Looks like the system ran out of memory while bring up the kernel. 
I'm cannot reproduce this issue with Tomato board's configuration plus CONFIG_NUC970_SD0_EN=y.
Can you check /proc/meminfo for free memory space after system booting with Tomato board's config?

VS0x01, 2021-04-22:
> Looks like the system ran out of memory while bring up the kernel.
> I'm cannot reproduce this issue with Tomato board's configuration plus CONFIG_NUC970_SD0_EN=y.
> Can you check /proc/meminfo for free memory space after system booting with Tomato board's config?

MemTotal:          60428 kB
MemFree:           12340 kB
Buffers:               0 kB
Cached:            43332 kB
SwapCached:            0 kB
Active:              816 kB
Inactive:          42684 kB
Active(anon):        168 kB
Inactive(anon):        0 kB
Active(file):        648 kB
Inactive(file):    42684 kB
Unevictable:           0 kB
Mlocked:               0 kB
SwapTotal:             0 kB
SwapFree:              0 kB
Dirty:                 0 kB
Writeback:             0 kB
AnonPages:           192 kB
Mapped:              472 kB
Shmem:                 0 kB
Slab:               3800 kB
SReclaimable:       2624 kB
SUnreclaim:         1176 kB
KernelStack:         208 kB
PageTables:           44 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:       30212 kB
Committed_AS:        780 kB
VmallocTotal:     958464 kB
VmallocUsed:         460 kB
VmallocChunk:     712372 kB

yachen, 2021-04-23:
This system uses huge amount of memory during booting. Did you you build your own filesystem? or are using using the one came long with our BSP?

VS0x01, 2021-04-23:
The board came with 2013 uboot and 3.10.101 kernel and it mounted SD card. 
I built 2016 uboot and this kernel using buildroot.
SD card has FAT filesystem. I just replaced uImage made with mkimage from uboot tools.

VS0x01, 2021-04-23:
[dmesg_3.10.101.log](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/6362088/dmesg_3.10.101.log)
Here is the old log.

yachen, 2021-04-26:
I still see lots of free space after system boot up.
Did you made any change except load tomato board's config plus SD0 support?
Was the SD card inserted during your test?
```
MemTotal:          60424 kB
MemFree:           47692 kB
Buffers:               0 kB
Cached:             8928 kB
SwapCached:            0 kB
Active:              788 kB
Inactive:           8308 kB
```

VS0x01, 2021-04-26:
> Did you made any change except load tomato board's config plus SD0 support?

No.

> Was the SD card inserted during your test?

Yes.

yachen, 2021-04-26:
Can the system boot up if you removed the SD card?

VS0x01, 2021-04-26:
I write uBoot to SPI and it reads and boots uImage from SD card...
I tried to boot kernel image via USB but it was hanging at _Calibrating loop delay_
So Idk how to boot w/out SD card.

yachen, 2021-04-27:
Hi, please try add lpj=744448 in bootargs in the dts file.

VS0x01, 2021-04-27:
I added, this place passed.
Now boot stay at _NET: Registered protocol family 1_

VS0x01, 2021-05-18:
I configured to compress rootfs and SD card driver works good, but there is still very little free memory.

closed 2021-05-18 by VS0x01