OpenNuvoton/NUC970_Linux_Kernel #67: how to enable PCM Interface in Sound

chandong83, 2021-04-26:
hello,

I would like to use i2s pcm interface mode.

do you have anything for reference?

and i would also like to know how to change from i2s mode to 24bit 384fs.

thank you

yachen, 2021-04-27:
Hi, Currently we don't have reference for PCM mode driver.

chandong83, 2021-04-27:
Hi, @yachen 
i am sorry i was also making inquiries from "NUC970_NonOS_BSP".
and is there any reference to I2S 16-bits on 24-bits frame?
thank you.

yachen, 2021-04-27:
Please try
1. Modify BITS_SELECT_16 to BITS_SELECT_24 in nuc970-i2s.c, and
2. Change 256 to 284 at line 109 in the same file.

chandong83, 2021-04-27:
@yachen 
Thank you for your reply.
in case this I2S(16-bits on 24-bits frame) left justified format is supported?

yachen, 2021-04-28:
Yes, FORMAT (I2S_CON[3]) control bit is used to select data format.

chandong83, 2021-04-28:
Hi, @yachen
i will check them. 
thank you

closed 2021-11-23 by ychuang3