OpenNuvoton/NUC970_Linux_Kernel #75: nuc977进入睡眠模式，但是不能通过按键唤醒设备

lifewifely, 2022-02-18:
1 我在驱动程序中开启了按键中断
      在porbe函数中，使用函数device_init_wakeup(&pdev->dev, 1)
2 通过SIMPLE_DEV_PM_OPS注册了key_drv_suspend函数，并在该函数里面通过enable_irq_wake(),使能了可以通过按键来唤醒系统
3 通过SIMPLE_DEV_PM_OPS注册了key_drv_resume函数，并在该函数里面通过disable_irq_wake(),关闭了可以通过按键来唤醒系统](url)
     SIMPLE_DEV_PM_OPS（wj_key_pm_ops, key_drv_suspend，key_drv_resume）
4 在driver中
        .pm = &wj_key_pm_ops

最后加载驱动

5 在应用程序中通过调用echo mem > /sys/power/state让设备进入低功耗状态，但是按下按键，系统没有恢复。

 @这种情况大概是哪里出现问题了呢？

如附件截图

lifewifely, 2022-02-18:
[image]

lifewifely, 2022-02-18:
[image]

lifewifely, 2022-02-18:
[image]

mjchen1, 2022-02-18:
which gpio are you using ?

lifewifely, 2022-02-18:
> which gpio are you using ?

hello ，i use PB11

lifewifely, 2022-02-18:
[低功耗附图.zip](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/8094926/default.zip)

lifewifely, 2022-02-18:
> which gpio are you using ?
hello，Attached is my program

低功耗附图.zip  this is my program

mjchen1, 2022-02-21:
只有gpio-eint的pin才能wakeup,
PB11不是gpio-eint pin,
可以參考device tree裡的gpio-eint pin,
pinctrl_gpio: gpio-eint {
		nuvoton,pins =
		<
		 7 0x0 0xF 0   /* eint0_H0 */
		 //5 0xB 0xF 0   /* eint0_F11 */
		 7 0x1 0xF 0   /* eint1_H1 */
		 //5 0xc 0xF 0   /* eint1_F12 */
		 7 0x2 0xF 0   /* eint2_H2 */
		 //5 0xD 0xF 0   /* eint2_F13 */
		 7 0x3 0xF 0   /* eint3_H3 */
		 //5 0xE 0xF 0   /* eint3_F14 */
		 //7 0x4 0xF 0   /* eint4_H4 */
		 //5 0xF 0xF 0   /* eint4_F15 */
		//7 0x5 0xF 0   /* eint5_H5 */
		 //6 0x5 0xF 0   /* eint5_G5 */
		 //7 0x6 0xF 0   /* eint6_H6 */
		 //8 0x1 0xF 0   /* eint6_I1 */
		 //7 0x7 0xF 0   /* eint7_H7 */
		 //8 0x2 0xF 0   /* eint7_I2 */
		 >;
};

lifewifely, 2022-02-21:
> 只有gpio-eint的pin才能wakeup, PB11不是gpio-eint pin, 可以參考device tree裡的gpio-eint pin, pinctrl_gpio: gpio-eint { nuvoton,pins = < 7 0x0 0xF 0 /* eint0_H0 _/ //5 0xB 0xF 0 /_ eint0_F11 _/ 7 0x1 0xF 0 /_ [...]

你好，我现在将按键脚改为了PH0脚，功能模式将 *gphmfpl |= （0x0f < 0） //将该ph0引脚设置成EINT0功能，也不能唤醒，你好，你们有一个在linux下中断唤醒的例子吗，或者有相关的参考吗，非常感谢。

mjchen1, 2022-02-21:
要先在menuconfig裡,enable "NUC970/N9H30 external I/O wakeup support"
[image]
可以參考drivers/input/keyboard/NUC970_keypad.c

lifewifely, 2022-02-22:
> 

1  嗯嗯，enable "NUC970/N9H30 external I/O wakeup support"这个选项我已经使能了。如下图：
[image]

[image]
[image]

2 我也按照你的建议，看了NUC970_keypad.c，里面我看到的有关唤醒的，是设置enable_irq_wake(IRQ_GPIO_START+NUC970_PH4+i);
[image]

3 我的程序也设置了，但是不行。
然后使能了唤醒设置，如下各种方式：
/*********************pin 使能中断唤醒***************************/
    enable_irq_wake(128+0xE0);
    enable_irq_wake(NUC970_PHH0);
    enable_irq_wake(which_irq[0]);

    irq_set_irq_wake((128+0xE0), 1);
    irq_set_irq_wake(NUC970_PHH0, 1);
    irq_set_irq_wake(which_irq[0], 1);
测试程序，都不能唤醒。

4 我看寄存器手册，我用PH0当做EINT0，将该寄存器低4位设置成1111，
 /*****************pin function set***************************/
    *pclken0 |= (1<<3);                     //APB Devices Clock Enable,这里用了
    //*gphmfpl &= ~(0x0f<<0);                //Multiple Function,PB11通用gpio功能 ,这里用了
    *gphmfpl |= (0x0f<<0);                //EINT0,外部唤醒中断功能，bit0-bit3都为1111
    *gphdir &= ~(0x1 << 0);                //我是0脚,PH0,方向为输入,这里用了
    *sys_wakeup |= (0x1<<0);  //bit 0
    *sys_wakeup |= (0x1<<25); //bit 25

5 你好，nuc977这个能不能在linux下进行休眠唤醒呢？如果能的话，我该怎么配置呢，或者说我现在胡配置哪里有问题呢，附件是我胡程序，能不能帮忙看一下呢，如果不能通过按键来唤醒系统胡话，我就不费时间在这上面胃，这个伙弄了几天了，我也参考网上其他的程序，都只需要简单胡配置，或许是我对nuc970系列胡芯片或者系统不熟悉，导致没有配置对，或者没有配置全。

lifewifely, 2022-02-22:
> 要先在menuconfig裡,enable "NUC970/N9H30 external I/O wakeup support" [image] 可以參考drivers/input/keyboard/NUC970_keypad.c
[程序.zip](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/8114230/default.zip)

这个是我的程序，帮我看一下我的配置，谢谢

mjchen1, 2022-02-23:
我在nuc977上,利用GPI_1(eint6),
看起來是可以wakeup的,
[image]
你可以先把 drivers/gpio/gpio-nuc970.c裡的eint interrupt enable如下圖的設定,
先確認你的板子是否可以正常喚醒系統
[image]

lifewifely, 2022-02-23:
> 我在nuc977上,利用GPI_1(eint6), 看起來是可以wakeup的, [image] 你可以先把 drivers/gpio/gpio-nuc970.c裡的eint interrupt enable如下圖的設定, 先確認你的板子是否可以正常喚醒系統 [image]

非常感谢你的回复
1 我按照你的建议去测试
   1.1 我打开PH0,PH1和PH7这个三个。 PI1以及PF11等相关io口用作其他的或者不太方便引出，我就没有测试。
[image]
[image]
   1.2 PH0,PH1和PH7这三个我测试的方式，直接按照你的方法打开，然后在终端上面按相应的按键，可以产生中断，如下图
[image]

   1.3 然后我写了一个应用程序，直接将系统进入到睡眠模式，此时按按键，还是不能唤醒。（这样直接让系统睡眠的测试方式应该是对的吧）
   1.4 这个是不是证明我的板子不能？
   1.5 我板子的按键原理图下：
[image]

2 你好
   2.1 你用的开发板是官方的开发板吗，还是其他型号的开发板
   2.2 我可以买一套你的开发板吗？买的话，我是直接找你买或者能不能给一个你用的开发板的型号和购买链接呢？
   2.3 你用的内核是这个NUC970_linux_kernel?你的.config能够发一份给我吗

3 真的非常感谢你这几天的回复，虽然到现在我也没有调通，真的非常非常感谢。我现在按照你的方式还是不能唤醒，就是证明我这个板子不能唤醒吗？

mjchen1, 2022-02-23:
原則上,能產生interrupt,應該就要可以wakeup系統,
但是我看你貼的,產生interrupt的圖,這張圖的最上面,有出現over-current的message,
我猜,這根pin有可能被設成usb相關的pin,
你可以檢查一下,其他的pin是否有一樣的情況.

附件是我所使用的.config file
[gpio_wakeup.zip](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/8128582/gpio_wakeup.zip)

lifewifely, 2022-02-25:
> 原則上,能產生interrupt,應該就要可以wakeup系統, 但是我看你貼的,產生interrupt的圖,這張圖的最上面,有出現over-current的message, 我猜,這根pin有可能被設成usb相關的pin, 你可以檢查一下,其他的pin是否有一樣的情況.
> 
> 附件是我所使用的.config file [gpio_wakeup.zip](https://github.com/OpenNuvoton/NUC970_Linux_Kernel/files/8128582/gpio_wakeup.zip)

你好，我弄通了，非常感谢你。
1 我开始没有通是因为用了PB11
2 后面改为PH1，还是没有通，原因是我在PB11的时候，以为是我的程序的问题或者是内核配置的问题，所以在内核配置的时候选择了如下图的
[image]
qi'shi
其实只需要选择第一项，就可以了

非常感谢你
另外请教一下SYS_nWAKEUP,这个引脚大概怎么使用来唤醒系统呢

mjchen1, 2022-02-28:
SYS_nWAKEUP是用來上電的,沒辦法用來使power down的系統wakeup