OpenNuvoton/NUC970_Linux_Kernel #77: HMI-N9H30 board kernel config

Barmallini, 2023-03-14:
Please, provide kernel konfig for this board.
With config from NUC970 kernel does not load (stuck)
[image]

ychuang3, 2023-03-14:
Hi,
Do you build the kernel with buildroot?
Please find buildroot: https://github.com/OpenNuvoton/NUC970_Buildroot
and select the defconfig 
$make nuvoton_n9h30_emwin_defconfig
$make

Barmallini, 2023-03-14:
Where did you do find nuvoton_n9h30_emwin_defconfig file?

Barmallini, 2023-03-14:
Actually, it starts ok when i copy /etc folder from board to target

ychuang3, 2023-03-15:
Looks like your image lost rootfs.
I still suggest to use buidlroot,and please just have a look at the README of NUC970_buildroot.
You can find nuvoton_n9h30_emwin_defconfig in configs folder. 
I guess you did not try "make nuvoton_n9h30_emwin_defconfig" as I have suggested.

Barmallini, 2023-03-16:
finally, i'm solve my troubles:
1. downloaded vmware image for nuc980
2. git clone https://github.com/OpenNuvoton/NUC970_Buildroot inside VM
3. make nuvoton_n9h30_emwin_defconfig
4. make

all builded without errors (except no networking support, ifup -a runs lo only)
Result: Linux version 3.10.108 (user@ubuntu) (gcc version 4.8.5 (Buildroot 2016.11.1-gfaa664b) )

buildroot for nuc980 have version of Linux 4.4, but does not have support for N9H30. how to merge configs?
image builded for NUC980 does not work on HMI-N9H30 board (no output to console)
Linux 3.1 is really old

ychuang3, 2023-03-16:
nuc980 linux 4.4 does not include any driver support for n9h30.  Not supported.
No plans about migrate n9h30 to latest kernel version in the near future.

Barmallini, 2023-03-16:
little offtopic then, is it possible to buy MA35D1 ?
if N9H30 out of support, what next?
what will be supported nearest 5 years and available to buy? NUC980?