OpenNuvoton/NUC970_Linux_Kernel #78: HMI-N9H30 board DTS file

Barmallini, 2023-03-14:
Where to get dts file for this board?

Barmallini, 2023-03-14:
How to enable device tree to make modification (add interrupts, gpio, interfaces, etc)

ychuang3, 2023-03-15:
Device tree enable/disable is configured from kernel menuconfig.
You can find related information by google search. It's not specific to n9h30.

Barmallini, 2023-03-16:
actually, it's specific
for example, Allwiner F1C200 (https://github.com/thirtythreeforty/businesscard-linux/blob/master/board/businesscard/v1/dts/businesscard.dts) for board
and
https://github.com/thirtythreeforty/linux/blob/f1c100s-v5.2.2/arch/arm/boot/dts/suniv-f1c100s.dtsi  for SOC

where to find N9H30.dtsi ?

ychuang3, 2023-03-16:
As of version 3.10, the device tree is not widely used.
Although NUC970 has support, it is recommended to use menuconfig only.
If you must use the device tree, you may face many trial and error. I am not sure.
The device tree file are in the arch/arm/boot/dts directory, which should be well known.
You can find 9h30-nudesign.dts here, please modify it based on your board configuration.
In addition, kernel menuconfig must open CONFIG_OF to enable device tree support..